iNavFlight/inav #3817: 3ch plane navigation - only rudder navigation

marikgl, 2018-08-28:
Hi,

I would like to use INAV on a 3ch plane.

Today, INAV uses Ailerons and Elevator in order to make a turn in a plane.
Bank it on a side with Ailerons and then pulls up Elevator.

In order to use INAV on 3ch plane, boat, rover - INAV should navigate by Rudder only.

Is there any way to get INAV navigate by rudder and not Ailerons?

Regards,
Mark

shellixyz, 2018-08-28:
I guess you could just mix the ailerons output to the rudder and pass the rudder channel from your RC to the ailerons channel.

marikgl, 2018-08-30:
It will always try to correct roll with rudder.

I need only navigate with the rudder.

closed 2018-11-12 by stale[bot]

hali9, 2018-11-21:
This is very interesting, can reopen this ?
> It will always try to correct roll with rudder.

This can be disable by set fw_turn_assist_yaw_gain=0, but I don't think it is not nesesery.
> I guess you could just mix the ailerons output to the rudder and pass the rudder channel from your RC to the ailerons channel.

Yes, but this wil work also in ANGLE mode and with the rudder itself, can not stabilize the plane.

Maybe set up posControl.rcAdjustment[YAW] by rollAdjustment and posControl.rcAdjustment[ROLL] = 0
in updatePositionHeadingController_FW
and then also set rcCommand[YAW] in applyFixedWingPitchRollThrottleController

Is it going to work well for the plane ?

SquashSquishSquaw, 2019-10-24:
I'm completely baffled.

Does this mean that iNav *cannot* fly an airplane that has only throttle, elevator and rudder?

Is there abolsutely no workaround for this!?? Such kind of airplanes are probably the most common airplanes one can buy, aren't they??