iNavFlight/inav #3818: OSD show wrong GPS coordinates

alterfettersack, 2018-08-28:
Hi in the INAV configurator SW the GPS reading is correct,
but in my  googles showed via OSD not.
It shows me an location an hour away...
Any hints or suggestions??

THX

shellixyz, 2018-08-28:
You are using 2.0 final ?

alterfettersack, 2018-08-28:
Jop...

# version
# INAV/OMNIBUSF4PRO 2.0.0 Jul 14 2018 / 22:40:31 (f2cf315ad)

alterfettersack, 2018-08-28:
screenshots:
[https://ibb.co/iVr8L9](url)
[https://ibb.co/i8FqYU](url)

alterfettersack, 2018-08-28:
screenshots:
https://ibb.co/iVr8L9
https://ibb.co/i8FqYU

shellixyz, 2018-08-28:
What are the good / displayed coordinates ?

alterfettersack, 2018-08-28:
This ones are correct: https://ibb.co/i8FqYU

Problem if i loose my plane and have still video i cant find it :/

THX

teckel12, 2018-08-28:
@alterfettersack So the latitude is correct but the longitude is incorrect?  Confused by the `X`.  Are you trying to hide your location or are you trying to highlight the latitude?

alterfettersack, 2018-08-28:
X is to hide my location, the problem is the longitude.
Latitude is correct.

fiam, 2018-08-28:
You’re running a release candidate. It’s a know bug and it was fixed in 2.0 final.

teckel12, 2018-08-28:
I wonder if setting `osd_coordinate_digits` to `8` if that would change anything?:

`set osd_coordinate_digits = 8`

alterfettersack, 2018-08-28:
FW updated, tested and now it works :D

Thank you very much !!!

fiam, 2018-10-27:
Closing, this was indeed fixed in 2.0.

closed 2018-10-27 by fiam