iNavFlight/inav #3819: Runcam Split 2  in iNav 2.0 bug.

lylus, 2018-08-29:
I tried to take pictures with the transmitter switch, but only if it was valid for the first time, and the transmitter switch was invalid after power failure and restart.

Flight control board:matek F722  

Runcam Split 2  in F722 TX4&RX4

advancetom, 2018-08-31:
Make a reboot over OSD and Check again

marcodcribeiro, 2018-09-03:
Tks, for the help.
So, if not a INAV issue i'll closeed this issue.

marcodcribeiro, 2018-09-03:
I try OSD reboot but dont work.

advancetom, 2018-09-03:
Did  you changed TX RX and give a try again?

lylus, 2018-09-03:
There is no problem with the wiring. It is effective to pull the remote control when the flight control is connected to the computer. However, if the computer connection is disconnected and the flight control is restarted, it is invalid.

bsdshneg, 2018-09-10:
This problem has been going on for a very long time!
The initialization of the UART takes place earlier than Runcam's time to load (if I'm not mistaken). This problem was promised to be solved in version 2, but the problem remained. If you reboot the controller from the OSD OSD menu, everything will work.

chan2328, 2018-09-13:
@lylus try to use RX3 & TX3.

putimir, 2018-10-06:
#2747

Nihilistic, 2018-12-12:
I'm having the same problem with my Runcam Split 2S wired to a Matek F405-CTR. I know the wiring and switches are set up correctly, because if I reboot the FC either via the iNAV configurator or OSD stick menu, all three of the transmitter switches will operate the Split 2S as they're supposed to. But, none of them work after the initial power up without a reboot of the FC with the Split 2S already powered up. I'm running iNAV 2.0.1

azolyoung, 2019-01-15:
Hi

Sorry to reply so late. Last week until now, it took some time to debug the issue that runcam split can't be controlled via iNav. This time, I finally reproduce the uncontrollable phenomenon. Because the Split data cannot be received correctly during the initialization process, the initialization fails. At present, I can't find the specific reason. Since I have spent a lot of time looking for the cause, but still can't locate the problem. So for the time being I made a compromise, using two cli commands to make the camera's features recognizable correctly.
The main changes to the test firmware linked below are:
1.Change the Split communication protocol to asynchronous, avoid blocking camera initialization, causing motor restart or other strange problems.
2.Added rcdevice_protocol_version and rcdevice_feature to manually identify the camera's characteristics and protocol version number.

If you still can't enable camera control using the following firmware, please paste the following three commands into the cli interface and try again:
set rcdevice_protocol_version = 1
set rcdevice_feature = 7
save

https://s3.amazonaws.com/rctmpfiles/inav_test_fws/inav_2_1_fix_rcsplit_telemetry_control/inav_2.1.0_MATEKF405.hex
https://s3.amazonaws.com/rctmpfiles/inav_test_fws/inav_2_1_fix_rcsplit_telemetry_control/inav_2.1.0_MATEKF405SE.hex
https://s3.amazonaws.com/rctmpfiles/inav_test_fws/inav_2_1_fix_rcsplit_telemetry_control/inav_2.1.0_MATEKF722.hex
https://s3.amazonaws.com/rctmpfiles/inav_test_fws/inav_2_1_fix_rcsplit_telemetry_control/inav_2.1.0_OMNIBUSF4PRO.hex
https://s3.amazonaws.com/rctmpfiles/inav_test_fws/inav_2_1_fix_rcsplit_telemetry_control/inav_2.1.0_OMNIBUSF4V3.hex

azolyoung, 2019-01-16:
Awesome! Thank you for your help. By the way, we would like to know the camera you used to test, is the Split 2, Split 2S or other camera ? And its firmware version?

azolyoung, 2019-01-16:
OK，Thank for the info.

Riffer, 2019-02-10:
I confirm, it works with F722 and a RunCam Split Mini V1 and the suggested parameters.
I can not configure the OSD in configurator, but actually the OSD works. Do I need another version though?

teckel12, 2019-02-11:
@azolyoung Could you supply the source or is it already merged into INAV?

azolyoung, 2019-02-11:
@teckel12  Yes, I've submit a pr  for this fixed,https://github.com/iNavFlight/inav/pull/4252, still waiting review

teckel12, 2019-02-11:
@azolyoung I have a RunCam Split 2S in the mail, I'll test the PR once it arrives.

Nihilistic, 2019-02-28:
@teckel12 Got your Split 2S yet? If so, any update on the status of this fix?

IvanKrys, 2019-03-02:
Hi!
I had the same issue. runcam was operating only after reboot of controller.
Hardware: Matek 405 CTR & Runam split 2 with latest firmware.
The Inav firmware in the links above along with the commands solved the issue.
As I can see this fix was not integrated into the latest release 2.1.0 as of Feb 25. this release does not recognize the commands stated above.

Alejandro-pucci, 2019-04-21:
Same issue on Matek F411-mini, inav 2.1 and Runcam split mini 2 FW: 2.3. No working controls after powering on, but after rebooting the FC everything seems to work on the bench.

teckel12, 2019-04-21:
@Nihilistic I do have my Split 2S up and running. But I haven't had much time (marathon season) and weather is just now cooperating. If I can remember what UART I used, I should try it out ;)

Nihilistic, 2019-05-13:
Oddly enough...the TX controls for the Split 2S work fine on my quad with the Matek F722-SE with iNAV 2.1 on it.

fiam, 2019-07-13:
Fixed by #4650

closed 2019-07-13 by fiam

ST02CMC, 2020-06-08:
> Hi
>
> Sorry to reply so late. Last week until now, it took some time to debug the issue that runcam split can't be controlled via iNav. This time, I finally reproduce the uncontrollable phenomenon. [...]

@azolyoung is there anychance that you can implement the runcam fix into Inav 2.5.0 for the 405 Wing please ?

digitalentity, 2020-06-09:
@ST02CMC what are the fixes you added to the test firmwares? Could you create a pull request so we can merge that?

ST02CMC, 2020-06-09:
> @ST02CMC what are the fixes you added to the test firmwares? Could you create a pull request so we can merge that?

@digitalentity Hi I did not create the fixes, @azolyoung found the solution, and built the versions listed above in 2.1.0, I was asked @azolyoung if he could add the fixes to version 2.5.0 and create a build for the Matek 405 Wing

digitalentity, 2020-06-09:
Cool! We plan to release a patch version 2.5.1 with some fixes, it would be great if @azolyoung can provide a fix for RCSplit as well.

ST02CMC, 2020-06-09:
> Cool! We plan to release a patch version 2.5.1 with some fixes, it would be great if @azolyoung can provide a fix for RCSplit as well.

@digitalentity just reading back on the post the 2 thing @azolyoung changed are as follows 

1.Change the Split communication protocol to asynchronous, avoid blocking camera initialization, causing motor restart or other strange problems.
2.Added rcdevice_protocol_version and rcdevice_feature to manually identify the camera's characteristics and protocol version number.

If that makes any sense to you as I am not a programmer 🤪

PR #4650 (2019-05-26): RunCam Device protocol updates
Supersedes #4252

Fixes #2747 
Fixes #3819

@Riffer, @Nihilistic, @ZinhoRacing, @Jetrell, @teckel12 - I don't have a RC Split with me right now, so I can't test these changes. Please, give this branch a try so we can get these fixes into 2.2