iNavFlight/inav #3821: gcc compile option -fno-common

robert-b, 2018-08-29:
when using the upper option there is a strange output:

Linking REVO
Memory region         Used Size  Region Size  %age Used
           FLASH:      237048 B       896 KB     25.84%
    FLASH_CONFIG:          0 GB       128 KB      0.00%
             RAM:       41372 B       128 KB     31.56%
             CCM:        7808 B        64 KB     11.91%
     BACKUP_SRAM:          0 GB         4 KB      0.00%
       MEMORY_B1:          0 GB         0 GB     -1.#Jobj/main/REVO/fc/cli.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/fc/config.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/fc/fc_core.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/fc/fc_init.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/fc/fc_tasks.o (symbol from plugin): In function `switchStates':
(.text+0x0): multiple definition of `switchStates'
obj/main/REVO/fc/fc_init.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/fc/fc_msp.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/fc/settings.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/flight/mixer.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/flight/servos.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/io/rcdevice_cam.o (symbol from plugin): In function `rcdeviceInit':
(.text+0x0): multiple definition of `switchStates'
obj/main/REVO/fc/fc_init.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/rx/eleres.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/blackbox/blackbox.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/cms/cms.o (symbol from plugin): In function `switchStates':
(.text+0x0): multiple definition of `switchStates'
obj/main/REVO/fc/fc_init.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/cms/cms_menu_ledstrip.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/io/ledstrip.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/telemetry/mavlink.o (symbol from plugin): In function `inputSource_e':
(.text+0x0): multiple definition of `inputSource_e'
obj/main/REVO/build/build_config.o (symbol from plugin):(.text+0x0): first defined here
obj/main/REVO/drivers/serial_usb_vcp.o (symbol from plugin): In function `usbVcpGetBaudRate':
(.text+0x0): multiple definition of `USB_OTG_dev'
obj/main/REVO/vcpf4/usbd_cdc_vcp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
%
make: *** [Makefile:336: obj/main/inav_REVO.elf] Fehler 1

when looking at the declaration of "switchStates" we find:

// used for unit test
rcdeviceSwitchState_t switchStates[BOXCAMERA3 - BOXCAMERA1 + 1];

but this is in the header file.
bad habit  - varaibles should never be declared in a header file.
which leads to multiple allocations of the variables.

the correct way to do this in the header file is:
// used for unit test
extern rcdeviceSwitchState_t switchStates[BOXCAMERA3 - BOXCAMERA1 + 1];

and the only declaration would be in the "C" file.

digitalentity, 2018-08-29:
How is this relevant to existing code?

robert-b, 2018-08-29:
-fno-common
In C code, this option controls the placement of global variables defined without an initializer, known as tentative definitions in the C standard. Tentative definitions are distinct from declarations of a variable with the extern keyword, which do not allocate storage.

Unix C compilers have traditionally allocated storage for uninitialized global variables in a common block. This allows the linker to resolve all tentative definitions of the same variable in different compilation units to the same object, or to a non-tentative definition. This is the behavior specified by -fcommon, and is the default for GCC on most targets. On the other hand, this behavior is not required by ISO C, and on some targets may carry a speed or code size penalty on variable references.

The -fno-common option specifies that the compiler should instead place uninitialized global variables in the data section of the object file. This inhibits the merging of tentative definitions by the linker so you get a multiple-definition error if the same variable is defined in more than one compilation unit. Compiling with -fno-common is useful on targets for which it provides better performance, or if you wish to verify that the program will work on other systems that always treat uninitialized variable definitions this way.

never, never, never declare variables in a header file!

digitalentity, 2018-08-29:
Hm... nice catch. Indeed we declare (likely as a mistake) a few vars in header files. We could benefit from making `-fno-common` a default.

digitalentity, 2018-08-29:
PR in place, thanks @robert-b

closed 2018-08-29 by digitalentity

robert-b, 2018-08-29:
that's causing me big problems in C++

robert-b, 2018-08-29:
all C++ problems solved.