iNavFlight/inav #3824: MATEK F405 MINI: Blackbox record is dangerous

lossUAVes, 2018-08-29:
## Current Behavior
When I start a recording in Flash chip with switch (assigned in modes), I arm motors with no airmode neither motors spining on arming, and when I start to fly the motors go crazy. The recorded archive seems corrupted and I can´t open it in blackbox viewer.

## Steps to Reproduce

1. Blackbox switch ON
2. Arm
3. Fly
4. Crash

## Expected behavior

Blackbox logging on flash chip with switch should work correctly on MATEK F405 mini with MATEKF405 firmware.

## Suggested solution(s)

I don´t know...

## Additional context

https://pastebin.com/THLhjWgH

- MATEK F405 Mini
-  INAV, version: 2.0.0

artusmuc, 2018-08-30:
Hello,

I have also a F405 Mini with activated blackbox. I have reduced to 50 % logging frame rate. 
(MAG-BARO-LED-SMARTPORT-IRCTRUMP-SBUS) 
I have only some problems with unbalanced motors - but i have no crash.

lossUAVes, 2018-09-02:
Ok, thanks. I will check my settings.

giacomo892, 2018-09-03:
I had the same issue on 2.0 with F405-CTR and a cheap Kingston class 10 sdcard.

The suggestion is to use a fast sdcard like a SanDisk A1 and reduce logging rate.

Still this needs to be investigated and solved.

DzikuVx, 2018-09-04:
Problem indeed seems to be related with the fact that gyro, looptime and SD card are working on high speed. Either lower looptime or sd card speed and test. Runnig blackbox on rates higher then 2kHz (1/2 rate in this case) does not makes much sense (we rather do not deal with frequencies above 1kHz). Even 1kHz (1/4 for this config) adds very little to the log.

lossUAVes, 2018-09-04:
I have slowed down to 1/32 and the quad is vibrating violently. I have tried to decrease the looptime and it also starts to vibrate. I imagine that this time it will be the PIDs. If under the looptime it seems that I have to refine the PIDs again. At the moment I still can not record data.

lossUAVes, 2018-09-05:
The file that I was able to record yesterday I opened it with blackbox viewer, but it will not let me open it with Plasmatree analyzer. Is it possible?. I downloaded an inav txt file and it seemed that analyzer also works with inav.

lossUAVes, 2018-09-05:
[blackbox_log_2018-09-04_232922.TXT](https://github.com/iNavFlight/inav/files/2352134/blackbox_log_2018-09-04_232922.TXT)

lossUAVes, 2018-11-01:
I have managed to use the blackbox and I have recorded some tests, but I can not open them with PlasmaTree Analyzer. I have also included axisD[2] to test but I still can not open them. I can not find anymore on the internet.

closed 2019-03-18 by stale[bot]