iNavFlight/inav #3825: PPM not working in iNav 2.0 any RC, working in 1.9.1

Nobatgeldi, 2018-08-29:
Board and Version
Matek F722 STD
INAV 2.0.0 stable release on Aug 20 2018
iNav 2.0.0 receiver
[image]
iNav 1.9.1 receiver
[image]

Behavior
PPM not working
Select PPM signal in Configuration, save and reboot, go to reciver tab

Tested and working in 1.9.1.

Could be configurator issue but I wouldn't speculate

shellixyz, 2018-08-29:
Confirmed, I can reproduce the issue. Tested with a IA6B receiver. Working with 1.9.1, not working with 2.0.0.

toff74, 2018-08-30:
it's the same problem for me

stronnag, 2018-08-30:
Works here (SPRF3, crappy Redcon DSMX RX).

shellixyz, 2018-08-30:
F7 only maybe then

tobimai, 2018-08-31:
Also not working on my Omnibus F3 Pro, using Spektrum AR610 receiver with PPM encoder. 
I will edit/delete the comment when I found a problem.

LucaBertuolo, 2018-09-05:
Same problem in Omnibus F3 AIO, switched back to 1.9.*

sniperxnl, 2018-09-13:
Also not working on F35

happyvalley-lmx, 2018-10-12:
Same problem in Omnibus F4 V3...

nanoperry, 2018-10-14:
when is the problem fixed we here nothing big problem i think

sramula, 2018-11-01:
Same problem on my Omnibus F4 PRO V2 (from Gearbest-maybe clone, but led strip works on dedicated conector). I use PPM encoder, on firmware OMNIBUSF4PRO v 1.9.1 signals works, on 2.0 there is no signal from reciever.

chinaling, 2018-11-02:
Same problem for me.
1.9 works, 2.0 does not show receiver when setting as PPM
using Matek F722 board firmware.

However my board is not a Matek board it's
https://www.aliexpress.com/item/F722-F7-Flight-Controller-With-OSD-Barometer-Built-in-5V-3A-BEC-For-RC-Drone-Support/32944307725.html?spm=2114.search0104.8.50.b3257c63CskxCU

digitalentity, 2018-11-18:
F7 PPM issues fixed in #4021

closed 2018-11-18 by digitalentity

chinaling, 2018-11-23:
Today I have reloaded Inav for M722 today using current stable versions.
 MultiWii API version received - 2.2.0
2018-11-23 @ 09:46:07 -- Flight controller info, identifier: INAV, version: 2.0.1
2018-11-23 @ 09:46:07 -- Running firmware released on: Oct 31 2018 19:27:11
2018-11-23 @ 09:46:08 -- Board: MKF7, version: 0

The release date is Oct 31 and the fix was reported as fixed on Nov 18.
This implies that the release is not available in the stable version.
I do not know if there is a later release of firmware or how it may be accessed.  I will be waiting until the next official release.

shellixyz, 2018-11-23:
Should be fixed in the development version. You can try this: 
[inav_2.1.0_MATEKF722.hex.zip](https://github.com/iNavFlight/inav/files/2611194/inav_2.1.0_MATEKF722.hex.zip)

SIM34ssimen, 2018-11-26:
> same problem i have matek f722 with ezuhf ppm reciever in 191 its working but no mixer in inav2.0
> when i put firmware 2.0 then no ppm but then i have mixer
> what must i do what am i doing wrong

I have 2 Radio controllers, one is Devention and another one is flysky. The flysky is working when using ppm.
, but Devo does not detected when using matek f722

giacomo892, 2018-11-26:
@SIM34ssimen have you tried the 2.1.0 firmware provided by @shellixyz few posts above?

SIM34ssimen, 2018-11-27:
yes I did and it works. Thank you very much.

sniperxnl, 2018-12-10:
I have a F35 with a EZUHF nano and it is still not working, is there a solution yet?

shellixyz, 2018-12-10:
@sniperxnl Please try this dev build (3dc3158ac) : [inav_2.1.0_FF_F35_LIGHTNING.hex.zip](https://github.com/iNavFlight/inav/files/2663318/inav_2.1.0_FF_F35_LIGHTNING.hex.zip)

SIM34ssimen, 2018-12-11:
On the computer, ppm rx was working. When the I took the quad out and try to arm, there was nothing happened. When I  use fw.1.9.1 it armed, I  switched back to 2.01 it still doesn't arm. Now I have to use 1.9.1firmware. Please fix this problem.

giacomo892, 2018-12-11:
@SIM34ssimen will be fixed in 2.1

walney-reptiles, 2019-05-05:
FC Matek-F411 ppm not working in 2.0.0 - 2.1.0 anyone got a fix for this?

lorenz4672, 2019-05-09:
Also same problem with omnibus F4 V3, ppm does not work at version 2.1.0.
Will there be a fix soon ?