iNavFlight/inav #3826: Auto disarm with rolled upside down

MichaelNevsew, 2018-08-29:
Hi guys,

Are there any way to automatically stop motors/disarm with my drone is rolled upside down. Basically, when it crashes.
Now when I crash (I fly with ALT HOLD mode) motors become rotating like I've pushed full throttle. Maybe some variable is responsible for the maximum roll/pitch degree after which it will disarm the quad?

Thanks in advance.

shellixyz, 2018-08-29:
It is not possible for the moment. The right way would be to disarm when a crash is detected. But not something as simple as disarm when angle > setting. It is something that is planned.

MichaelNevsew, 2018-08-30:
I teach children to fly and they oftenly don't switch to disarm when they crash due to some kind of panic. that's why I thought to use auto-disarm or failsafe or whatever to save motors from throttling when on the ground.

closed 2018-11-12 by stale[bot]