iNavFlight/inav #3827: Automatic determining of magnetic declination doesnt work on fw 1.8 - 2.0 in Poshold cruise mode

DiscoMan18, 2018-08-30:
450 size quadcopter, matek f405-ctr, radiolink ts100 mini (same behaviour earlier with Naza 2 GPS module) 

## Current Behaviour

With inav_auto_mag_decl = ON when you try to fly straight forward in POSHOLD CRUISE mode using only PITCH command, the quadcopter doesn't fly to the previously chosen direction, and takes about 10 degrees to the left. If you set inav_auto_mag_decl to OFF and manually set your current magnetic declination via configurator or CLI (about 10.2 deg in my case), it flies as expected, i.e straight forward.  
Also when hovering in place if you try to give it sudden ROLL input, the quadcopter swings not only in ROLL axis, but makes a complex diagonal movement. This behaviour first was noticed starting from fw 1.8.1 till the very latest 2.0 release. 

## Steps to Reproduce

1. set  inav_auto_mag_decl = ON, enable POSHOLD and user control mode CRUISE
2. fly straight forward using only PITCH command
3. watch how the quadcopter takes left or right, depending on your mag declination

## Expected behavior

Straight forward flight with maintaining previously chosen direction

## Additional context

[LOG00085.TXT.zip](https://github.com/iNavFlight/inav/files/2335545/LOG00085.TXT.zip)

# version
# INAV/MATEKF405 2.0.0 Aug 20 2018 / 19:17:50 (dbdd1656a)

# reset configuration to default settings
defaults noreboot

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# map
map TAER

# serial
serial 2 1024 115200 38400 0 115200
serial 3 2 115200 38400 0 115200
serial 4 16 115200 38400 1200 115200

# led

# color

# mode_color

# aux
aux 0 1 0 900 2100
aux 1 3 0 1450 2000
aux 2 9 0 1800 2000
aux 3 18 0 900 1150
aux 4 29 1 1775 2000
aux 5 30 1 1050 1200
aux 6 31 2 1825 2100

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 25 4 V
osd_layout 0 1 1 1 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 6 12 12 V
osd_layout 0 7 13 1 V
osd_layout 0 9 1 7 V
osd_layout 0 11 7 1 V
osd_layout 0 12 1 3 V
osd_layout 0 13 24 1 V
osd_layout 0 14 24 3 V
osd_layout 0 15 24 2 V
osd_layout 0 19 1 2 V
osd_layout 0 20 18 13 V
osd_layout 0 21 1 13 V
osd_layout 0 22 14 3 V
osd_layout 0 23 18 1 V
osd_layout 0 24 13 2 V
osd_layout 0 26 24 9 V
osd_layout 0 28 23 11 H
osd_layout 0 30 1 14 V
osd_layout 0 33 1 8 V
osd_layout 0 34 11 4 H
osd_layout 0 35 1 4 V
osd_layout 0 40 18 2 V
osd_layout 0 41 24 7 V
osd_layout 0 42 24 8 V
osd_layout 0 46 22 9 H
osd_layout 0 47 22 10 H
osd_layout 0 50 11 4 V
osd_layout 0 55 1 10 V

# master
set acc_hardware = MPU6000
set acczero_x = 69
set acczero_y = -11
set acczero_z = -183
set accgain_x = 4083
set accgain_y = 4102
set accgain_z = 4062
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set mag_declination = 1020
set magzero_x = -138
set magzero_y = 13
set magzero_z = 91
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 8
set min_throttle = 1060
set max_throttle = 1992
set failsafe_delay = 10
set failsafe_off_delay = 0
set failsafe_throttle = 1275
set failsafe_procedure = RTH
set current_meter_scale = 185
set yaw_jump_prevention_limit = 150
set model_preview_type = 3
set small_angle = 180
set disarm_kill_switch = OFF
set gps_min_sats = 5
set deadband = 7
set alt_hold_deadband = 20
set inav_auto_mag_decl = OFF
set nav_use_midthr_for_althold = ON
set nav_extra_arming_safety = OFF
set nav_user_control_mode = CRUISE
set nav_wp_radius = 200
set nav_auto_speed = 1000
set nav_auto_climb_rate = 250
set nav_manual_speed = 1000
set nav_manual_climb_rate = 500
set nav_emerg_landing_speed = 130
set nav_rth_altitude = 4000
set nav_mc_bank_angle = 25
set nav_mc_hover_thr = 1400
set osd_video_system = PAL
set osd_time_alarm = 120
set osd_alt_alarm = 1000
set osd_dist_alarm = 50000
set i2c_speed = 100KHZ

# profile
profile 1

set mc_p_pitch = 43
set mc_i_pitch = 14
set mc_d_pitch = 41
set mc_p_roll = 38
set mc_i_roll = 14
set mc_d_roll = 41
set mc_p_yaw = 80
set mc_i_yaw = 35
set yaw_p_limit = 250
set nav_mc_vel_z_i = 60
set nav_mc_pos_xy_p = 20
set nav_mc_vel_xy_p = 35
set nav_mc_vel_xy_d = 70
set nav_mc_heading_p = 70
set thr_mid = 40
set thr_expo = 30
set rc_expo = 60
set rc_yaw_expo = 35
set roll_rate = 30
set pitch_rate = 30
set yaw_rate = 15

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 250
set vbat_warning_cell_voltage = 270

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

# 
---

INAV/MATEKF405 2.0.0 Aug 20 2018 / 19:17:50 (dbdd1656a)

shellixyz, 2019-05-15:
@DiscoMan18 You might want to test #4709. We're going to test if it is working as intended and will make a test build for you

DiscoMan18, 2019-05-15:
Thanks, will do that in a couple of days, will wait for the test build!

shellixyz, 2019-05-26:
The changes has been integrated into 2.2-RC1. You might want to directly test it. The declination tables have been updated.

closed 2019-08-08 by stale[bot]