iNavFlight/inav #3829: Be able to use PWM 2 pins for something else if model has just 1 motor

GenuineFlollie, 2018-08-31:
## Current Behavior
The motor 2 PWM output pins cannot be configured to be used for something else (like a servo)

## Desired Behavior
Be able to assign PWM 2 pins to an RC Channel (for custom servo or other funky stuff)

## Suggested Solution
If the second slot under Motor Mixer is empty, start a new servo mixer rule that default drives PWM 2, or additionally add a combo box in the row to allow selection between PWM 2 or PWM 5 and up.

## Who does this impact? Who is this for?
As most flying wings and airplanes don't have 2 motors anyone with 1 motor can do something cool with PWM 2, currently only 3 of 4 outputs are usable for most people.

## Additional context
Personally i wanted to do this: https://www.instructables.com/id/Convert-a-Servo-to-switch-on-and-off-LEDs/

Best regards,

Floris

Tygrys-1, 2018-09-03:
It may be a problem, as the motor designed outputs frequently share the same timer. So if the motor ESC is working on 400 Hz or 1 kHz the other motor output will also be driven with that frequency. That way it could not be used for the standard 50 Hz PWM servo.

On some FC's this might not be the case, but as a design rule it is made that way. You may check the target source (target.c) for your board to see the timer mapings.

GenuineFlollie, 2018-09-03:
Ah, ok I didn't know that :)
I solved my problem with using the PWM outputs of the receiver,  this is however not a solution if you have an SBUS only receiver. 
Anyways, tx

closed 2018-09-03 by GenuineFlollie