iNavFlight/inav #3836: Flaps on a fixed with using Inav2 and a Matek 405 Wing

DonaldHillNZ, 2018-09-03:
Hi, I've spent considerable time trying to get flaps working with Inav 2 and a new Matek 405 Wing FC.
I've used S7 as recommended somewhere and mapped to Ch5 on the Taranis x9d plus. Despite being able to see the receiver working on ch5 in Inav for ages I couldn't get any actual physical movement of the servo. After searching for ages I came across someone suggesting that you need to turn on Flaperon in the modes to make it work. It did work, however, I only had a very tiny flap movement. Finally overcame this by using MAX as the source. However, when I use the slow up and slow down functions for the servo they both revert to delay up and delay down and not slow. I've checked to make sure the functions work correctly when not connected to the 405 Wing and Inav. I'm not even sure I am using the correct way in Inav for flaps but I've searched the internet for hours and can find nothing. Id really appreciate some info as well as whether this is a bug or not? Also any info on how to use a 3POS for the flaps? So far only works on one switch position.

Donald

Pairan, 2018-09-03:
would be interesting for me too, as my next project uses Matek F405-WING and Flaps!

DonaldHillNZ, 2018-09-03:
Hi Pairan, I've been building planes etc for 40 odd years and please don't get me wrong but I think the 405 Wing with Inav is truly exciting. However, I am completely stumped by the flaps issue. I've also built successfully 4 racing drones as well so I'm no newcomer. Hopefully, I'm missing something completely. Definitely get a Matek 405 Wing regardless.

MATEKSYS, 2018-09-03:
https://www.youtube.com/watch?v=_VL57d8IYnE&t=18s

davidngrc, 2018-09-04:
@DonaldHillNZ yes, bypass the FC, connect your flag servo directly to the X8R pwm pin
I suggest you bind your x8r receiver in ch 9-16 mode, that means the pwm in x8r is map to channel 9, 10, 11 to 16 (you can still use the sbus, get the same channel 1 to 16)

DonaldHillNZ, 2018-09-04:
Thank you so much. That explains a lot. Ill give it a go in the morning. 
Ive got the GPS with compass, oled, all working well and just the vtx 
and cam to setup. Then put it all together and go flying. Great board 
and Inav.

Donald

<mention@noreply.github.com>
Sent: 4/09/2018 4:39:20 PM
Matek 405 Wing (#3836)

shellixyz, 2018-09-04:
You can just mix your flaps channel to an output. No need to connect them directly onto your receiver. Just add a servo mix rule.

DonaldHillNZ, 2018-09-04:
Hi Shellixyz. Yes you can but it does not work without really messing around with setups and no slow up or down when connected to the Matek F405 Wing using Sbus. 
SOLVED though by using davidngrc's recommendation. Simply bind the X8R in mode 4 is using D16 and channels 9 -16 as PWM jumping 1&2 and 3&4. Remember that the channels are 9 to 16 when remapping. Everything works that way with the F405 on Sbus and the flaps on their own PWM channel. Slow up and down works as well.

jgilbertfpv, 2018-09-11:
speaking of flaps, could this work on a flying wing with two elevons per side? so the inner elevon goes down and outer goes up? https://www.youtube.com/watch?v=qQI_1P7voCg

DonaldHillNZ, 2018-09-16:
We are just in the middle of shifting house so might not be able to be online for a while but the only way I could solve this for my situation was to use the PWM outputs. I didn't use the Flaperon in the modes when doing this. trivejb I think you need to choose Has Flaps in the configuration in Inav to have that option. I think it's in the configuration tab but can't check right now. Still haven't solved the jittering with all other servos. It's quite bad.....

trivejb, 2018-09-17:
Thanks for the response. Using CLI I SET HAS-FLAPS=ON, set a mode for FLAPERONS, in Mix Tab I have S5 and S7 set to FLAPS Receiver tab shows channel is active.  but still do not get any servo movement. I need to do a little more tuning because the FLAPERONS mode would not show active (color green) when switch activated it. Could be part of the problem... Still looking... Starting to make my ET Vector units better all the time :).
Hope your move goes well and thanks again