iNavFlight/inav #3837: Add climb and descent angle in degrees for winged craft

loflyfpv, 2018-09-03:
Would it be possible to have an OSD option for the climb and descent angle expressed as degrees positive or negative to zero (level), in addition to the visual and numeric options already implemented?

teckel12, 2018-09-03:
@loflyfpv I know it's possible as the pitch value is already exposed. Just a matter of attaching the value to a new OSD option. I'm surprised it doesn't already exist actually.

loflyfpv, 2018-09-03:
@teckel12 I'm guessing the conversion from pitch value to degrees would be simple too.

digitalentity, 2018-09-03:
Pitch on OSD is expressed in degrees.

loflyfpv, 2018-09-04:
@digitalentity i'll check that this evening. i must confess that i've never used the pitch, roll and yaw in the osd and didn't think to try them before posting.

closed 2018-11-17 by stale[bot]