iNavFlight/inav #3842: Switch disarm takes ~1.3s instead of 0.25s

krzysztofmatula, 2018-09-05:
Inav 2.0, OMNIBUSF4V3.

I use the default switch disarm delay (250ms) and it works like that on the bench.
On the field, after flying, the delay is much longer (but not always...).

I wonder what could be taking the additional ~1s and why...?

See this video showing ~1.3s delay:
https://www.dropbox.com/s/p3chcw41qvasrsi/disarm-delay.MOV?dl=0

My diff:
```
# version
# INAV/OMNIBUSF4V3 2.0.0 Aug 20 2018 / 19:10:22 (dbdd1656a)

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature -TX_PROF_SEL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ON_USB

# map

# serial
serial 0 2 115200 115200 0 115200
serial 5 64 115200 38400 0 115200
serial 30 16 115200 38400 57600 115200

# led
led 0 15,15:S:C:2
led 1 14,15:S:CW:2
led 2 9,15:S:CW:2
led 3 8,15:S:CW:2
led 4 7,15:S:CW:2
led 5 6,15:S:CW:2
led 6 1,15:S:CW:2
led 7 0,15:S:C:2
led 8 11,0:N:C:1
led 9 10,0:N:CW:1
led 10 9,0:N:CW:1
led 11 8,0:N:CW:1
led 12 7,0:N:CW:1
led 13 6,0:N:CW:1
led 14 5,0:N:CW:1
led 15 4,0:N:C:1

# color

# mode_color

# aux
aux 0 0 5 1500 2100
aux 1 2 1 900 1200
aux 2 20 1 1200 1400
aux 3 4 1 900 1075
aux 4 9 0 1250 1500
aux 5 8 0 1500 2100
aux 6 11 1 1500 2100
aux 7 12 2 1700 2100
aux 8 32 2 1300 1700
aux 9 33 2 1700 2100

# adjrange

# rxrange
rxrange 0 1019 2019
rxrange 1 1019 2019
rxrange 2 1019 2019

# osd_layout
osd_layout 0 0 8 1 V
osd_layout 0 1 24 1 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 15 14 V
osd_layout 0 9 1 2 H
osd_layout 0 11 25 2 V
osd_layout 0 12 9 14 V
osd_layout 0 13 1 7 V
osd_layout 0 14 0 1 V
osd_layout 0 15 24 7 V
osd_layout 0 19 25 3 V
osd_layout 0 20 16 14 H
osd_layout 0 21 4 14 H
osd_layout 0 22 14 1 V
osd_layout 0 23 16 1 V
osd_layout 0 28 21 14 V
osd_layout 0 29 24 1 H
osd_layout 0 31 0 2 V
osd_layout 0 34 10 1 H
osd_layout 0 40 0 3 V
osd_layout 0 53 1 14 H
osd_layout 0 54 2 14 V
osd_layout 0 76 1 2 H
osd_layout 1 1 22 11 V
osd_layout 1 20 2 9 V
osd_layout 1 21 2 8 V
osd_layout 1 23 2 11 V
osd_layout 1 28 22 9 V

# master
set looptime = 500
set gyro_sync = ON
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 100
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 140
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 70
set acc_hardware = MPU6000
set acczero_x = 79
set acczero_y = -32
set acczero_z = -27
set accgain_x = 4067
set accgain_y = 4097
set accgain_z = 4000
set mag_hardware = HMC5883
set magzero_x = 22
set magzero_y = -82
set magzero_z = -71
set align_mag_roll = 1800
set align_mag_yaw = 450
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set rssi_channel = 9
set serialrx_provider = SBUS
set blackbox_rate_denom = 2
set min_throttle = 1050
set max_throttle = 1950
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set current_meter_scale = 270
set model_preview_type = 3
set deadband = 0
set nav_extra_arming_safety = OFF
set nav_user_control_mode = CRUISE
set nav_auto_speed = 600
set nav_manual_climb_rate = 500
set nav_rth_altitude = 2000
set nav_mc_hover_thr = 1268
set osd_video_system = PAL
set async_mode = GYRO
set name = KM
set stats = ON
set stats_total_time = 39907
set stats_total_dist = 348219
set stats_total_energy = 1745688

# profile
profile 1

set heading_hold_rate_limit = 50
set thr_mid = 35
set thr_expo = 30
set rc_yaw_expo = 70
set roll_rate = 80
set pitch_rate = 80
set yaw_rate = 80

# battery_profile
battery_profile 1

set vbat_cell_detect_voltage = 440
set vbat_max_cell_voltage = 440
```

giacomo892, 2018-09-06:
Could it be due to having stats enabled? Can you try without them enabled?

krzysztofmatula, 2018-09-06:
You're right, that's stats!
And on the bench it also can be seen, but arming state must be enabled for longer than MIN_FLIGHT_TIME_TO_RECORD_STATS_S (10s).

giacomo892, 2018-09-06:
We need to optimize that, making it disarm and then taking care of the stats

krzysztofmatula, 2018-09-06:
I agree. I'm looking for best place to execute the flash write.

closed 2019-02-02 by stale[bot]