iNavFlight/inav #3844: Temperature SENSOR

Kryptooo, 2018-09-06:
Please ad a uart temperature sensor for Longrange flights with lion batteries.

giacomo892, 2018-09-07:
Do you have some kind of hardware to suggest?

leeph, 2018-09-07:
I was thinking of this the other day. It would be logical to use the same sort of temperature probes used in Spektrum telemetry, EagleTree telemetry etc as these are available everywhere and for cheap. It wouldn't make any sense to use up a digital UART for this, just an ADC pad would do I should think.

teckel12, 2018-09-07:
I would think an i2c temp sensor would be better. They're more accurate than an analog sensor that needs calibration and using i2c means you wouldn't use a dedicated port (just share the i2c with whatever). I've done this with Arduino and it's super straight forward and cheap.

shellixyz, 2018-11-28:
I bought some KM75A sensors. Cheap and seems like a good solution for on board temperature measurements. Up to 8 can be used on the same bus, 2.8 to 5.5V power supply, temperature range from -55°C to 125°C with 0.125°C resolution. I will try to add support for it

wx4cb, 2018-11-28:
@shellixyz another popular one which can be had REALLY cheap is the venerable DS18B20. I know it's not I2C but it only requires 1 pin (outside of power/Ground) and can be purchased in a TO92 case which makes them perfect for sticking in small places

https://cdn-shop.adafruit.com/datasheets/DS18B20.pdf
https://www.adafruit.com/product/374

digitalentity, 2018-11-28:
Probably 1-wire is even easier to implement using an UART - https://www.maximintegrated.com/en/app-notes/index.mvp/id/214

wx4cb, 2018-11-28:
@digitalentity REALLY easy to implement and the good thing is, those DS1820B's can be parasitically powered so you literally only need, data and ground

shellixyz, 2018-11-28:
@wx4cb It is an interesting sensor. Not as cheap as the LM75A but the possibility to only use 2 wires in total to connect the sensor is really nice. We would have to implement the 1-wire protocol though.

shellixyz, 2018-11-28:
@digitalentity We could also add support for the DS2482 I²C bus master

wx4cb, 2018-11-28:
most of the work has already been done: https://stm32f4-discovery.net/2014/05/13-reading-temperature-with-dallas-ds18b20-on-stm32f429-discovery-board/

digitalentity, 2018-11-28:
@shellixyz it would be nice to see DS2482 put on the FC itself 😄 
@wx4cb implementing it for INAV would require considerably more work

shellixyz, 2018-11-28:
@digitalentity Indeed. Suggestion for Matek for their next FC ? :smile:

wx4cb, 2018-11-28:
@digitalentity I realise that, if i was a better coder with more understanding of STM32s i could help lol and i agree a bus master would be nice on board. no need to worry about - oh crap how do i connect this LOL

sniperxnl, 2018-12-12:
Any idear how far it is?
I hope it will work with the MCP9808

shellixyz, 2018-12-12:
@sniperxnl I'll start working on it when I get the sensors. I bought some LM75 and DS18B20. They have been shipped but I'm still waiting for them. If you want to contribute some MCP9808 chips I can add support for it too.

darrindevine, 2018-12-20:
@shellixyz was searching and stumbled on this feature. I'm brand new to iNav. I think it's an important addition. Look forward to your work!

kardon18, 2019-01-28:
> 
> 
> @shellixyz  I'll start working on it when I get the sensors. I bought some LM75 and DS18B20. They have been shipped but I'm still waiting for them. If you want to contribute some MCP9808 chips I can add support for it too.

I look forward to the support release for the DS18B20 Dallas sensor.

Constructively, the LM75, being a SOIC 8. Needs to be board mounted. As most are.
This makes for a rather large temp sensor for most application. Its a bit hard to just slip it under the heat shrink on the VTX or ESC.  
While mounting it directly the VTX RF shield or the ESC heatshink; reduces thermal dissipation efficiency on those devices. 

Any reason that an NTC Thermistor isn't considered to connect to a spare ADC?
The ET Vector uses the Thermister approach. I have used may different cheap 10K NTC thermisters with it. And I've tested the Vectors calibration with multiple digital thermometers, and have found the output reading doesn't fluctuate more than a few degrees from actual.

shellixyz, 2019-01-29:
@kardon18 

> Constructively, the LM75, being a SOIC 8. Needs to be board mounted. As most are. This makes for a rather large temp sensor for most application.

That's why the DS18B20 in TO92 package is nice. And only 3 wires needed instead of 4 for the LM75. For some applications using a LM75 is fine though. The PCB doesn't need to be larger than 10x10mm. Would be fine for measuring the air or battery temperature. But as you said not ideal for measuring the motor or VTX temperature for example.

> Any reason that an NTC Thermistor isn't considered to connect to a spare ADC?

It would be possible and would be pretty easy. The downsides are that there are often not a lot of analog inputs on the FCs and electrical interference can have a big impact on analog signal. Also you need one signal wire per sensor so if you want say 4 sensors then you need 2 wires for power and 4 wires for signal (and also 4 resistors or 4 caps). 6 wires in total. The DS18B20 has the disadvantage to need an interface chip but it is very small and can be put close to the FC then you only need 3 wires total for the bus whatever the number of sensors used.

PR #4282 (2019-02-08): Temp sensors improvements
Closes #3844 

* Add 1-Wire support through the DS2482 IF chip (new driver)
* Add driver for the DS18B20 1-Wire temperature sensor
* Support for multiple LM75 sensors
* Support for the DS18B20 sensors
* Add CLI command `temp_sensor` to configure sensors
* Logging of all temperature sensor values
* Display of all temperature sensor values on the OSD with support for alarms
* New MSP messages to query and set sensor configuration and read temperatures
* Enable 1-Wire and DS18B20 temp sensors on all F4 and F7 targets
* Up to 8 temperature sensors supported

Syntax of the [... cut]
closed 2019-02-25 by shellixyz