iNavFlight/inav #3845: Servos in fixed wing not smooth

DonaldHillNZ, 2018-09-06:
Hi, I've set up the Inav and the Matek 405 Wing and all is working well so far including GPS with a Beitian which was a breeze and I'm using flaps as suggested in channel 13 after rebinding my X8r RX. All good however all the servos connected to the F405 are not at all smooth. They stutter is the best description. They operate ok with full deflection in each direction but are jerky in their operation. Is there a way in Inav to sort this out maybe with PIDS which I'm not familiar enough with to go diving in without some help. By the way, they are digital servos.

GeoffO63, 2018-09-07:
Yes I'm having the same issue....

DonaldHillNZ, 2018-09-07:
Despite trying for ages I cant overcome it

DonaldHillNZ, 2018-09-07:
Got to be an issue with either the F405 Wing or that in combination with Inav.

GeoffO63, 2018-09-07:
Thanks for the advice.  I just tried that now and no difference I'm afraid..

digitalentity, 2018-09-07:
Ok, this is odd. I have an F405Wing straight from Matek and no issues with
jittery servos. Were did you buy your board?

GeoffO63, 2018-09-07:
Recently from Hobby King. 

I thought it may have been the hinges sticking but are moving freely when the control rod is disconnected.

atomiclama, 2018-09-07:
Hmm I'm just starting with wings and iNav, My servos are not running as smooth as I was expecting. Thought it was just my cheapo servos or to great expectations. Maybe something else. Thanks for the tip about the cap. I'll try that and do some checks on the PWM signals.

ShikOfTheRa, 2018-09-07:
If not power, try 50hz servo rate?

I'm running 2.0 on wing in 2 planes and haven't noticed servo jitter - but am running 50hz as older servos in plane

DonaldHillNZ, 2018-09-08:
Sorry I've been away for a few days. I already had added a 1000mu to the battery power and as a thought also added one to the esc power. No difference. I purchased my board off Banggood. Are these not genuine Matek?? When I get back home tomorrow I'll try the 50Hz on the servo rate. If anything the messing around I've done has made it worse but can see no reason why. As a check, I ran three servos through channels 9 to 12 and kept my flaps on 13. All as smooth as they should be.

GeoffO63, 2018-09-09:
I'm using Corona CS-939MG analog servos.
iNav set to 50Hz by default.

GeoffO63, 2018-10-07:
Just checking back in on this issue.

Donald, what brand and model of servo are you using?

I'm going to try some others.

Geoff

DonaldHillNZ, 2018-10-08:
Hi Geoff,

I think they are a generic no name brand in a model purchased from HK. 
I,m using it as a test bed.

However Ive tried a few different brands such as Corona, Hextronic etc. 
It has made no difference what brand Ive used.

Ive had to put a short halt on working on this since we are in the 
middle of a big move to a different house.

Should be back onto it in a couple of weeks.

Donald

<author@noreply.github.com>
Sent: 8/10/2018 11:54:29 AM

GeoffO63, 2018-10-08:
Hi Donald,

Thanks for your response.

I flew the wing today and the jitty servos weren't noticable whilst flying.

I think I'll just leave it as is for the moment.

Good luck with the move.

Geoff

DonaldHillNZ, 2018-10-08:
Well I think I might just do the same to. Let me know if you have any 
problems later on.

Just want to get this move over. Get back to my flying etc.

Believe it or not we "have" to take around 100 pot plants. Grr

All 20 odd planes and my model making equipment already gone.

Regards

Donald

Sent: 8/10/2018 6:50:13 PM

thomasekdahlN, 2018-10-30:
I have the same problem.

R9 Slim+ receiver with FLEX FW and F.PORT
Matek F405 Wing
Inav 2.0 - F.Port

Running iNAV in MANUAL mode (No gyro/acc help on servos) and the servos are very jerky/not smooth in their movement.

Anything I can do to help you developers to debug the problem?

vicdzen, 2018-11-13:
The same jitter here. Matek F405-CTR.
I have noticed that jitter is related to PID loop frequency (more frequency - less noticeable jitter).
I think it is performance (CPU load) related. It is also related to OLED screen feature (if on - more noticeable jitter). In terms of "more noticeable jitter" I mean the servo position refresh rate became less than 5Hz. For example if I throw roll stick from extreme left to extreme right quickly, servo arms make several (5-10) noticeable stops during their full travel. During normal steering you can clearly hear that 5-10Hz stuttering.

DonaldHillNZ, 2018-11-13:
Still sorting out our recent move but will be back looking at this problem shortly. The question I have is that is this problem being looked at in Inav or a combination of Inav and the Matek F405-CTR?

vicdzen, 2018-11-13:
Can’t say, only one FC under inav. CTR and other Mateks are pretty the same thing so I think it’s common problem. It doesn’t affect stab and nav modes noticeably in the air, that’s why there are not so much reports on that, but it does if you know how it should be in manual, especially during fast acro moves. There is delay in servo responce.

digitalentity, 2018-11-13:
Still needs confirming on my end. Will look into this

GeoffO63, 2018-11-15:
> The same jitter here. Matek F405-CTR.
> I have noticed that jitter is related to PID loop frequency (more frequency - less noticeable jitter).
> I think it is performance (CPU load) related. It is also related to OLED screen feature (if on - more noticeable jitter). In terms of "more noticeable jitter" I mean the servo position refresh rate became less than 5Hz. For example if I throw roll stick from extreme left to extreme right quickly, servo arms make several (5-10) noticeable stops during their full travel. During normal steering you can clearly hear that 5-10Hz stuttering.
> PS. Servos are not the from cheap range. While connected directly to receiver - smooth as silk.

This makes sense given I'm using an OLED screen. 
When I get a chance I'll disable the screen to check and report back.

Geoff

DonaldHillNZ, 2018-11-29:
Hi all. I can confirm that in my case once the OLED screen feature is disabled the jittering problem disappears. Would be nice if this could be fixed since I was using the OLED. I've been unable to check until now since we have had a major move of house.......

Donald

digitalentity, 2018-11-29:
OLED-related issues can't be fixed. OLED update is quite time-consuming and during the update servos are stuck.

vicdzen, 2018-11-29:
Strange actually. If CPU is not 100% loaded (it is not now) there is a chance to squeeze some time for LCD.

digitalentity, 2018-11-29:
The problem is that we need that ~10ms time in one chunk. Scheduling OLED to spread the update is not worth the effort because it's not updated in flight anyway.

DonaldHillNZ, 2018-11-29:
I fully understand. The OLED was a nice to have but that's OK. Easily live without it. Its good to know what was causing it and that it can be remedied by choosing off.

vicdzen, 2018-11-29:
So, if it doesn’t cause jitter and doesn’t update screen inflight I’m ok with that jitter on the ground. Thanks for clarification!

digitalentity, 2018-11-30:
I think this can be closed now.

closed 2018-11-30 by digitalentity

ArchitectJuan, 2019-06-09:
I was having the same issue but because my negative wires were connected to different points on the PDB it started twitching. So I paired all my negative and positive (5v) together and soldered them to the PDB 5v and negative and it stopped twitching. I'm doing a wing FYI.
Hope that helps someone.

M0j0Risin, 2022-03-01:
Just to add to this in March of 2022...  I ran into the same issue this morning on INAV 4.1 after installing an OLED screen on one of my fixed wing models. That is, the servos were noticeably stuttering when the OLED screen was attached and went back to normal when it was removed. However, increasing the global I2C speed to 800 KHz on the configuration tab seems to have solved the problem or at least reduced it to an acceptable level. The I2C peripherals on my model are this generic little DIYMall OLED screen and a Matek ASPD-4525 airspeed sensor which seem to still be playing nicely together at that speed. I do not use a magnetometer on this model, but I assume it would still work fine since the INAV tooltip notes specify they recommend 800 KHz for multirotors which are much more likely to utilize a magnetometer in the first place. I should also mention that this is on a Matek F765-WSE flight controller so I can't really say if this advice would apply to older boards. Anyways, happy flying.

ronnygrimm, 2024-05-18:
Hi 
Fixed this by disabling telemetry!
I had the same problem with stuttering servos in fixed wing with MATEK F722wing with DJI controller 2 and Dji air unit 1
Fixed by disabling the telemetry in INAV config other features. Now servos are running smooth 
Probably overload on CPU with OLED or Telemetry enabled