iNavFlight/inav #3848: Disarm delay

oldrootbeer, 2018-09-08:
Hi. Whilst bench testing my motors, I went to disarm. I didn't hold the switch for long enough, and quite rightly, it did not disarm. However, I have blackbox set to record on the arm switch. The blackbox data suggests that there was no recording when the disarm was enabled briefly.
Does this mean that other modes, e.g. turn assist etc will briefly disable if the arm / disarm switch is accidentally pressed for a short while?
Obviously, while ambling around taking photos, then 0.2 seconds wont make a difference. But to someone who is 'kickin the split s', could be a problem.

digitalentity, 2018-09-08:
If you have some other modes set up on the same switch as arming (which is generally a bad idea) - they will disable before machine is actually disarmed.

oldrootbeer, 2018-09-08:
Thank you. So its a bad idea to have other modes on the arm switch. I have angle, turn assist and blackbox. I will look into a better way...

closed 2018-09-08 by oldrootbeer