iNavFlight/inav #3853: Led_strip interferes with voltage and current reporting

access3424, 2018-09-10:
## Current Behavior

It seems that whenever I turn on led_strip my board doesn't not report battery voltage and current. Everything is configured correctly and the Leds are working fine. When I disable led_strip vbat and current are reported as normal on inav configurator and  s.port telemetry.

CLI:
# diff

# version
# INAV/KAKUTEF4V2 2.0.0 Aug 19 2018 / 21:14:39 (dbdd1656a)

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature -TX_PROF_SEL
feature THR_VBAT_COMP
feature VBAT
feature GPS
feature CURRENT_METER
feature LED_STRIP
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 32 115200 38400 0 115200
serial 3 2 115200 115200 0 115200

# led
led 0 6,11::CB:1
led 1 7,11::CB:1
led 2 8,11::CB:1
led 3 8,12::CB:1
led 4 7,12::CB:1
led 5 6,12::CB:1
led 6 6,13::CB:1
led 7 7,13::CB:1
led 8 8,13::CB:1
led 9 8,14::CB:1
led 10 7,14::CB:1
led 11 6,14::CB:1
led 12 1,11::CN:2
led 13 2,11::CN:2
led 14 2,12::CN:2
led 15 1,12::CN:2
led 16 1,13::CN:2
led 17 2,13::CN:2
led 18 11,11::CN:2
led 19 12,11::CN:2
led 20 12,12::CN:2
led 21 11,12::CN:2
led 22 11,13::CN:2
led 23 12,13::CN:2

# color

# mode_color

# aux
aux 0 0 0 1750 2100
aux 1 1 1 1300 2100
aux 2 4 1 1700 2100
aux 3 9 1 1700 2100
aux 4 8 2 1875 2100
aux 5 21 2 1425 1575
aux 6 12 2 900 1175

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 13 1 V
osd_layout 0 1 1 12 V
osd_layout 0 4 8 6 H
osd_layout 0 7 17 11 V
osd_layout 0 9 1 2 H
osd_layout 0 11 24 12 V
osd_layout 0 12 17 12 V
osd_layout 0 13 4 6 V
osd_layout 0 14 21 2 V
osd_layout 0 15 21 6 V
osd_layout 0 22 17 3 V
osd_layout 0 23 10 3 V
osd_layout 0 28 21 1 V
osd_layout 0 30 1 8 V
osd_layout 0 31 21 3 H
osd_layout 0 32 2 8 H
osd_layout 0 34 10 10 V
osd_layout 0 38 2 10 V
osd_layout 0 40 2 2 V
osd_layout 0 46 2 2 H
osd_layout 0 48 8 12 V
osd_layout 0 49 23 7 H
osd_layout 0 53 1 11 V
osd_layout 1 45 0 0 V

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 70
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 43
set acc_hardware = MPU6500
set acczero_x = 27
set acczero_y = -9
set acczero_z = -8
set accgain_x = 4092
set accgain_y = 4093
set accgain_z = 4018
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 81
set magzero_y = -35
set magzero_z = 37
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 8
set blackbox_rate_denom = 5
set blackbox_device = SPIFLASH
set min_throttle = 1035
set motor_pwm_rate = 8000
set motor_pwm_protocol = MULTISHOT
set failsafe_procedure = RTH
set current_meter_scale = 290
set model_preview_type = 3
set switch_disarm_delay = 100
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set nav_auto_speed = 600
set nav_manual_speed = 600
set nav_land_slowdown_minalt = 125
set nav_land_slowdown_maxalt = 550
set nav_min_rth_distance = 700
set nav_rth_altitude = 1500
set nav_mc_hover_thr = 1450
set smartport_uart_unidir = ON
set osd_rssi_alarm = 40
set osd_time_alarm = 8
set name = BEHEMOTH

# profile
profile 1

set mc_p_pitch = 75
set mc_i_pitch = 50
set mc_d_pitch = 30
set mc_p_roll = 75
set mc_i_roll = 50
set mc_d_roll = 30
set mc_p_yaw = 95
set mc_p_level = 27
set max_angle_inclination_rll = 450
set max_angle_inclination_pit = 600
set dterm_notch_hz = 125
set dterm_notch_cutoff = 90
set rate_accel_limit_roll_pitch = 3600
set rate_accel_limit_yaw = 720
set heading_hold_rate_limit = 30
set roll_rate = 55
set pitch_rate = 55
set yaw_rate = 25

# battery_profile
battery_profile 1

set bat_cells = 4
set battery_capacity = 1300
set battery_capacity_warning = 390
set battery_capacity_critical = 260

- FC Board name and vendor: Hollybro Kakute F4V2
- INAV version string: 2.0

simtcrom, 2018-09-10:
I also have reported [#3851](
https://github.com/iNavFlight/inav/issues/3851) for led strip issue.
My issue is that turn indicator not working properly only right and down works.
Can you try if you have left and up working?

digitalentity, 2018-09-10:
Both ADC and LESTRIP timer use `DMA2_Stream4`. Once we finish https://github.com/iNavFlight/inav/pull/3833 we can look into possible remap options.

access3424, 2018-09-24:
@szpero did you modify the Hex or you can do this from the CLI?

Szpero, 2018-09-24:
@access3424 Modified target and compiled. CLI doesn't allow to move DMA.

Now I'm looking at target.h for KAKUTE and I think that LED handling should have deepers split between V1 and V2. Target.c has correct pinout for V1 and V2.
KAKUTEF4V2 PC8 DMA2_Stream2
KAKUTEF4 PA1 DMA1_Stream4

access3424, 2018-09-24:
@szpero Could you please share the hex because I can’t modify it myself!

digitalentity, 2018-09-24:
Already fixed in #3833.
KAKUTEF4V2 will use DMA2_ST2
KAKUTEF4 will use DMA1_ST4
```

access3424, 2018-10-01:
How do i modify the rousource list to show motor 5 and 6? It only shows motors 1-4

digitalentity, 2019-01-05:
Fxied by #3833

closed 2019-01-05 by digitalentity