iNavFlight/inav #3858: Autolaunch throttle by stick position

putimir, 2018-09-13:
It would be useful to have the ability to set the autolaunch throttle by stick position;

If using different types of batteries (3,4,5S, lipo, liion...), or launching from various scenarios you often have to change launch throttle to have an optimum launch (in order not to have too much torque roll, but enough to have safe takeoff).

I suppose we could have a setting in CLI like `nav_fw_launch_thr_by_stick` ( = 1), and autolaunch would use throttle set by stick. Before launch is detected, we would of course still use nav_fw_launch_idle_thr

Thanks

edlabh, 2018-10-23:
Great idea

FPVZaphod, 2018-11-20:
I love the simplicity to not choose the throttle value every time I launch. Wouldn't this be nice to include it into the battery profiles? Autolaunch throttle connected to the Auto-Profile function, so it changes the trottle according to the battery recognized.

putimir, 2018-11-20:
Yeah but, it's not just battery related; it's often other factors: wind, terrain, testing new props etc... 

Well, that's my idea / wish - but it seems it's not that interesting to other folks...so I don't know

vavisimo, 2018-11-20:
+1 interested too in such option

shellixyz, 2018-11-20:
Launch mode throttle in the battery profile is a good idea. Cruise throttle and pitch2thr should also be added to it. And why not an option for manual override of the launch throttle even if that could be somewhat dangerous if you reduce the throttle too much for the asked pitch but I'm not against it.

FPVZaphod, 2018-11-21:
The final launch mode activation happens when you raise the throttle by more than 50%. How will you manage to get a launch throttle of less than 50%? I guess such a manual throttle override should be only possible with Launch Mode on Switch, and not permanent enabled Launch Mode?

Arvycka, 2018-11-21:
Launch mode is working perfectly at the moment. I don't see any problem of launching the craft manually once in a while. Adding launch profiles will further add to the complexity of iNav where it is not strictly necessary and there will be more settings for user to remember to set which will defeat the purpose of fool-proofing the feature by the said settings.

putimir, 2018-11-21:
As far as fool-proofing goes: as I wrote in the first post, default behavior shouldn't and wouldn't be touched, instead, an extra cli parameter would be introduced in order for an "advanced" pilot to have the option to "knowingly" customize launch parameters and thus be aware of the caveats of this action.

As for the 50% throttle: think anything less than 50% would "take" the config-set autolaunch anything more than 50% would be used as custom throttle (I don't have wings that I autolaunch with less than 50% throttle)

But, as it seems that idea the stirs a lot of controversy and not a lot of interest...so...

shellixyz, 2018-11-21:
> The final launch mode activation happens when you raise the throttle by more than 50%

That's not how it works. You don't need to raise the throttle to more than 50% for the launch to work, only over `min_throttle`. Anyway the idea was just to maybe add a setting like `nav_launch_manu_thr` or `nav_launch_thr_mode`.

Avegawanderer, 2019-04-22:
> Launch mode throttle in the battery profile is a good idea. Cruise throttle and pitch2thr should also be added to it. And why not an option for manual override of the launch throttle even if that [...]

I would love to have launch throttle linked to battery profile, as well as cruise throttle and pitch2thr, just like suggested above. Manual setting by throttle stick would be nice too, having a setting like nav_launch_manu_thr. It would be also great to see battery profile-related settings in one place in the configurator.

breadoven, 2022-11-26:
Closed by https://github.com/iNavFlight/inav/pull/8134.

closed 2022-11-26 by breadoven