iNavFlight/inav #3860: Airspeed sensors not working

pilatus11, 2018-09-14:
## Current Behavior

Both digital and analogue airspeed sensors not working when looptime is above 2k. When set up on 4k using analogue via ADC 3 I was getting 60kmph on bench and with the digital sensor via I2c it would read 145kph on the bench.  Reduced looptime to 1k or 2k both worked perfect.

In the above example I was testing on a 2 x  Matek f405 wing flight controllers and omnibus F4v3.

Tested on both 1.9.1 and 2.0 stable.

## Steps to Reproduce

1. self-explanatory
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

Tim2309, 2018-09-14:
I can confirm the bug with an i2C sensor ms4525
The higher the loop is set the bigger is the problem.
With 8k the sensor shows around 11500 in the sensor tab on the workbench.

digitalentity, 2018-11-02:
@Tim2309 can you please check https://github.com/iNavFlight/inav/pull/3977 ?

Tim2309, 2018-11-07:
with 2.1.0 and 8K I got good results