iNavFlight/inav #3863: I get a different current value in OSD and S.Port Telemetry and this gives me different mah used in osd and on the taranis. why is this?

access3424, 2018-09-14:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)
* [IRC channel](irc://irc.freenode.net/#inavflight)
* [Slack channel](https://publicslack.com/slacks/inavflight/invites/new)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

wx4cb, 2018-10-31:
maybe related tot his?

https://github.com/iNavFlight/inav/issues/3853

shellixyz, 2019-02-02:
Please test 2.1.0-RC2 and report whether the issue is still present or not

closed 2019-04-17 by stale[bot]