iNavFlight/inav #3865: Unable to choose higher than 7 for servo in mixer FT405 wing

nosredna000, 2018-09-15:
[image]

## Steps to Reproduce

1.  External PWM Servo Driver ON
2.  Mixer tab will not allow assigning of servo higher than 7.  FT405 wing has 9 S9

## Expected behavior
Unable to select anything higher than 7

## Suggested solution(s)
Show me how I'm dumb and looking over the right check box lol :)

## Additional context

CLI dump
https://pastebin.com/Ai0iACfX
---

- FC Board name and vendor: FT405 
- INAV version string: # INAV/MATEKF405SE 2.0.0 Aug 20 2018 / 19:22:34 (dbdd1656a)

Tygrys-1, 2018-09-15:
You may try that in CLI
It might be a configurator problem.

nosredna000, 2018-09-15:
Could u give me a quick how to? I tried a few things before making my post but was unable to get anything.  Sorry for being vague bc im on mobile.

nosredna000, 2018-09-17:
Got it figured out.. the servo naming was confusing me.. Starting with Servo0 etc.

closed 2018-10-05 by DzikuVx