iNavFlight/inav #3866: External Compass Support on SpeedyBee AIO using i2c not working

Painless360, 2018-09-16:
I'm building a model with the SpeedyBee AIO flight controller and iNav 2.0, latest release. 

The external compass is the all-in-one M8N GPS Compass module that also contains a compass. I've used this type of GPS/compass module on all of my builds (fixed wing and multi-rotor) without issue on iNav 1.9 and below. 

Configuration of the GPS is automatic but the external compass (HMC5883L type) shows as red and won't connect. I've checked the wiring (SCL to SCL and continuity etc.) and it's all perfect. I've also swapped the GPS until and have the same issue. Also tried the AUTO and HMC5883 settings in the configurator with no luck. Also tried removing the compass setting and powering the GPS before and after setting the compass type. No luck. 

Is this is a known problem? Looks like the i2c isn't doing anything..

datamann, 2018-09-16:
I use Beitian BN-880 and it works great. Maybe you need pull up resistors on yours.

Painless360, 2018-09-16:
Interesting. Thanks for the reply. To be clear - you're also using a SpeedyBee AIO board?

datamann, 2018-09-16:
Btw
For those of you who have a SpeedyBee AIO FC and don't want to use UARTs or inverters for Smart Port and Smart Audio, I have compiled a new firmware where I use S5 and S7 (motor ports on fc) as soft serial. No inverters needed or anything just solder signal to the pads and configure in INAV. Of course you can't use S5 and S7 for motors...

Painless360, 2018-09-16:
Thanks for the heads up Datamann. That could come in handy as I could do with another for Smartaudio. I've ordered that other GPS you're using to try to check and then a note can be added to the Wiki for that FC. I'll test and come back to you.

Painless360, 2018-09-27:
I've tried an BN-880 compass unit and no luck. Scoping the pins on the board I can't see any signals from SDA/SCL at all so I'm starting to suspect my hardware here if you have it working..

datamann, 2018-09-27:
Hmm.. strange..  as you say maybe you have a hardware issue... I wonder if the problem is only the pads or circuit. Maybe it's possible to route I2C to other pads.... I will try to figure that out...

Painless360, 2018-10-24:
It's not hardware - in a last effort to figure out where the problem lay I've flashed the board with Betaflight 3.5.2 and it works perfectly - https://www.dropbox.com/s/t6pa1h9x8duqldr/2018-10-24%2008.50.58%20-%20Copy.jpg?dl=0

Looks like something in the interaction with this FC and iNav?

datamann, 2018-11-04:
Hi @Painless360 ,
INAV 2.0.1 is now released and if I'm not totally mistaken this issue is fixed.
Regards
Stig Sivertsen

datamann, 2018-11-05:
Did you try to activate:
No reboot sequence
Manual baud rate - Try default value first and set to 38440 if that dosent work

callagga, 2018-11-05:
Thanks. Well.... I hadn't pushed the DFU button....but in my defence I have never had to push it before for the original iNav flash, or whenever doing flashing on the same board with betaflight(?)

Painless360, 2018-11-06:
Hi Datamann, flashed and I can confirm it's sorted - thank you. What was the cause?

datamann, 2018-11-06:
Hi @Painless360 ,
I was not the one hwo fixed it but as far I know INAV didn't ACK all messages from the GPS so the I2C definitions was changed..

Ref.
https://github.com/iNavFlight/inav/pull/3797

Regards
Stig Siversten

datamann, 2018-11-06:
@Painless360 

If it works could you please close this issue?