iNavFlight/inav #3867: No telemetry in F.Port mode on MATEK F722

CcLaFleche, 2018-09-16:
# dump

# version
# INAV/MATEKF722 2.0.0 Aug 20 2018 / 19:21:31 (dbdd1656a)

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature THR_VBAT_COMP
feature VBAT
feature TX_PROF_SEL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
[... 850 lines cut ...]
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 3
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 360
set battery_capacity = 6300
set battery_capacity_warning = 1575
set battery_capacity_critical = 630
set battery_capacity_unit = MAH

rb1205, 2018-09-20:
Let's cover the basics: have you deleted and rescanned the telemetry sensors after switching from s.port to Fport? Does the telemetry work if you flash betaflight? Have you tried different UARTS? did you check you connected the s.port wire to the tx pin?

CcLaFleche, 2018-09-20:
Good settings, no problem on all UARTs, works under BF.
No need to re-scan the telemetry sensors because I am under FrOs and works well under BF.

rb1205, 2018-09-24:
I just tested my brand new matek f7, FPort working telemetry included without issues. R9 slim with latest Flex firmware installed, S.Port (white) wire going into TX2 pin, Taranis X9D with opentx 2.2.2. Relevant diff settings follows:

serial 1 64 115200 38400 0 115200
set rssi_channel = 16
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON

CcLaFleche, 2018-10-21:
Fport works fine, but rssi is 50%.
In sensors rssi is 100%...

rb1205, 2018-10-22:
Have you set rssi_channel to 16? in your provided config it isn't.