iNavFlight/inav #3869: The airplane would'nt hold the altitude of  "nav_rth_altitude" after engage into RTH mode

87040109qq, 2018-09-17:
## Current Behavior

The airplane would'nt hold the altitude of  "nav_rth_altitude"(nav_rth_home_altitude set to zero;nav_rth_altitude set to 100M),it only hold about 60M altitude!when hover on the home position. 

## Steps to Reproduce

1.After launched the airplane for 5-6s ,I switched into RTH mode immediately. The plane come back and hover on the home ,but not hold the altitude at nav_rth_altitude(100M) .Seems like hold the altitude at current hight(60M).

## Expected behavior

Should'nt the plane  climb/descend to 100M after reaching home if landing is not enabled. Set nav_rth_home_altitude  to 0 to stay at nav_rth_altitude (100M)

## Additional context

---

- FC Board name and vendor:
OMNIBUSF4V3 
- INAV version string:
INAV 2.0 stable

87040109qq, 2018-09-17:
Have anyone meet the same problem?or that is normal for 2.0!

Szpero, 2018-09-24:
Same here - INAV 2.0, FIXED alt_mode, 75m. Plane was climbing to 60m and it was its cruise altitude to base.

closed 2018-12-07 by stale[bot]