iNavFlight/inav #3870: OMNIBUS F4 Pro Corner, UART 3 don't work with telemetry

Frank-Seider, 2018-09-17:
I have GPS on UART 1 and S-Bus on UART 6 and it works, but the UART 3 which I want to use for telemetry do nothing. Is it because I2C is on UART 3? I try to use Softserial but no success. What can I do?

teckel12, 2018-09-17:
@Frank-Seider It's my understanding the pro corners has issues with inverted SBUS/S.Port and I2C, making it mostly useless for most INAV purposes.

Frank-Seider, 2018-09-17:
What do you think about the firework V2 FC in this context?

Tygrys-1, 2018-09-17:
Omnibus F4 shares UART3 with I2C and maybe this is the case. Do you have the compass enabled in FC configuration? Or other I2C device?
I don't know the logic INav uses to determine I2C/UART3 choice.

teckel12, 2018-09-17:
@Frank-Seider All I know is that the pro corners FC has some issues and doesn't work well at all with INAV. I don't think any developers ever had this board to make it fully compatible with INAV. And then there's the I2C issue that seems to not be fixable. Best to avoid this board with INAV.

teckel12, 2018-09-17:
@Frank-Seider I always suggest getting a REAL Omnibus F4 Pro FC. I've never had any issues as long as it's not purchased from China. I have no experience with the firework V2.

Frank-Seider, 2018-09-17:
@teckel12: The firework is a FC from airbot. This FC is listed in the latest inav version and has 4 UART's. I hope ist works... 
@Tygrys-1 : No, the compass is not enabled. Is there any possibility, to disable this i2C sharing on UART 3?

teckel12, 2018-09-17:
@Frank-Seider Just because there's a HEX file a FC doesn't mean that everything works or works the way you expect.  This is why I stick with the Omnibus F4 Pro FC.  Everything works and works as expected.

I'm going to try using the Matek F722 / FCHUB-6S PDB combo at some point as it seems to be mature.  I can't think of another FC I'd even mess with at this point.

closed 2018-12-02 by stale[bot]