iNavFlight/inav #3872: Motor 4 not working on Omnibus F4 V6

thibv, 2018-09-19:
## Current Behavior
I cannot start motor 4 on my Omnibus F4 v6 board

## Steps to Reproduce

1. Flash Omnibus F4 V6 board with iNav 2.0.0
2. Go to Motor tab
3. Try to activate motor number 4

## Expected behavior
Motor 4 should work with iNav 2.0, board Omnibus F4 V6 flashed with OMNIBUSF4 firmware

## Suggested solution(s)
Is it possible to have a OMNIBUSF4FW version for this board, like in Betaflight ?

thibv, 2018-09-19:
Information from this page : https://store.myairbot.com/flight-controller/omnibus-f3-f4/omnibusf4v6.html
I need to flash these board with FIREWORKSV2 firmware.

closed 2018-09-19 by thibv

CaptainYags, 2018-10-08:
Did you get it to work with the fireworksv2 target?  I can't get my iBus receiver to be recognized

teckel12, 2018-10-10:
@thibv How's the v6 different than the v4 and v5 other than marketing nonsense?

thibv, 2018-10-11:
I've found the page, it's from the UnmannedTechShop page :

> The V6 omnibus has quite a few changes:
New firmware target (OMNIBUSF4FW) - after flashing go to CLI and type set gyro_to_use = second , and save to get the IMU working.
5x UART Ports
5V 1A and 8V BEC for FPV camera
Current source jumper selecter (see manual for more information)
The V5 omnibus includes a current source selection jumper (see pinout below for more details), also the omnibus f4 v5 uses a new 8 pin Sh 1.0 Socket to make connecting a 4in1 ESC much easier. This is compatible with up to 6S batteries. 
The V4 omnibus main feature is that the solder jumper to selected between PPM/SBUS is no longer needed.  The board also features some minor hardware revisions to increase the boards robustness even more.
The V3 fixes some of the LDO problems of the V2, so now you can power this board directly from your battery (up to 4S Lipo)