iNavFlight/inav #3873: Omnibus F4 Will Not Power On.

Hopalonger, 2018-09-20:
Was Setting Up an Omnibus F4 Pro and got the Receiver Working then powered off and came back the receiver was not detected. I rebooted the Board and plugged in the lipo to make sure the board had power. It worked fine then During a reboot Inav Said Something along the line of "Fatal Error, Can not proceed with Reboot". Now the FC does not show any signs of life. I checked the regulators and they are producing the 3v3 that is expected and close to 5v. Any Suggestions on what I should do to fix this? If not what about a replacement FC for cheap?

teckel12, 2018-09-20:
@Hopalonger You can't brick it, just reflash the firmware.

Hopalonger, 2018-09-20:
@teckel12 The board has no Led and does not show up in device manager like it used to.

teckel12, 2018-09-20:
@Hopalonger Ah! Sounds like something popped.

shellixyz, 2018-11-19:
Looks like a hardware failure, closing

closed 2018-11-19 by shellixyz