iNavFlight/inav #3874: Airbot Omnibus aio f4 v6 ppm not working

tjeerd750, 2018-09-20:
The omnibus AIO f4 v6 has no controllable inverter on the sbus pin.  Ppm input won't work on this pin. I tried other pins but there is no signal. After contact with airbot they say there is no ppm pin. The information on the airbot site is wrong. Is there any way the firmware can be modified to use a ppm receiver? I am using fireworksv2 as target. I can't use sbus with my turnigy 9x with dht module. This board has 6 uart, there should be one that works.

teckel12, 2018-09-20:
@tjeerd750 If it's an F4 board there must be a physical inverter for SBUS. Therefore, it's possible (with some hardware modifications) to remove the inverter. But it may just be easier to get another FC.

tjeerd750, 2018-09-20:
Is it not possible to use another uart for ppm? In that case I will buy an sbus receiver.

CaptainYags, 2018-10-08:
Were you able to get SBUS running?  I'm having trouble getting mine going with the fireworksv2 target

tjeerd750, 2018-10-12:
Sbus receiver not working either, guess we are stuck with useless boards. I did get an email from airbot that PPM should work on the PB8 pin, shared with scl.

digitalentity, 2018-10-14:
@tjeerd750 can you post a link to the board so I can buy one and troubleshoot?

tjeerd750, 2018-10-15:
Hi Konstantin,

This is the board I bought:
https://store.myairbot.com/flight-controller/omnibusf4v6.html

Thank you for the support,
I hope you can get it working.

teckel12, 2018-10-15:
@tjeerd750 Have you tried connecting your FC to a different UART? The UART for SBUS is inverted so it won't with non-SBUS.

digitalentity, 2018-10-20:
Ok, I'm waiting for the OMNIBUS AIO F4 V6 board to arrive in a few days - then I'll be able to debug.

gustavkavac, 2018-11-24:
hello. I'm using a omnibus f4 v6 board for a wing. Everything seems to work fine except that it doesn't get my xm+ sbus signal. I read something about that the softserial is messing with the first uart so it has to be disabled, but i have no clue how to do that. I've been using betaflight for a long time, but Inav is different and i can't find how to disable or enable softserial. Is that the cpu based serial port slider. If so, i already have that disabled in the configurator. Is there a way to disable that through CLI? I don't know what to do because Sbus needs a hardware inverter, but i believe that this fc only has this one, which is on uart 1. What should i do? I can use any other pin. All i ineed is sbus.  

Bord: omnibus f4 v6

# version
# INAV/OMNIBUSF4 2.0.1 Oct 31 2018 / 19:07:40 (a04e68aaa)

tjeerd750, 2018-11-24:
Hi,

I could only get it to work with inav 1.9, with other versions I can't get
sbus signal working.

minerran1, 2018-12-07:
I can't get this to work either.  The documentation Airbot provided appears to be wrong.  Which uart is reserved for SBUS, Uart6 or Uart2 or Uart1?   From the resource table they provided vs. the board silkscreen showing pads, it seems like it could be any of these.   My receiver is paired with my taranis and I had this working on a Naze32.  thanks

Asizon, 2018-12-16:
Airbot provided documentation is wrong for now, we are going to correct it, for now you can try the new target i have created on Betaflight for OmnibusF4 V6, SBUS pad have a inversor, and assigned uart is UART6.
https://github.com/betaflight/betaflight/pull/7129

Asizon, 2019-01-06:
@digitalentity on OmnibusF4 V6 PB8(PPM) correspond to SCL named pin,can you try it please?

digitalentity, 2019-01-06:
OMNIBUS F4 V6 is FIREWORKSV2 target. There is PPM in pin defined on PB8 (and in schematics IIRC), but since it has an inverter, it may be tricky to get working.

digitalentity, 2019-01-06:
This is a minor difference from FIREWORKSV2 - such small difference should be handled as a target variant, not a separate target to avoid bloat. Closing as this is mainly about adding a new target which is discussed in #4006

closed 2019-01-06 by digitalentity

Asizon, 2019-01-06:
Yes, need to make OMNIBUSF4FW variant,same as Betaflight OMNIBUSF4V6 target.I think its neccessary also for Inav plataform

minerran1, 2019-01-07:
> 
> 
> For rx ppm input RX1 should be work,SBUS pin with building inversor is only for sbus

Asizon,  you say rx ppm "should work" but did you test it and prove that it does in fact work?

My experience with this board is that its a complete disaster.  sbus didn't work for me, nor did the gps and I2C magnetometer on the BN-880 I connected.  Airbot has errors in their documentation such as specifying the same uart being used for two different things simultaneously.   Using an AnyFC F7 I purchased from Banggood, I was able to get everything working (had to solder wires to add the MS5611 barometer)  So I know the problems I had on the airbot board were not due to any external cause.

Asizon, 2019-01-08:
Hi @minerran1
sbus works perfectly in uart6.The problem is in the incomplete firmware for it. In Betaflight we have already integrated it and we make changes and updates about it, I2C scl and sda work in Betaflight