iNavFlight/inav #3875: No GPS. Matek CTR Matek GPS SAM M8

bookworm1707, 2018-09-20:
I am used to working with betafligt and cleanflight on racing quads (No GPS). 

I have a new flight controller and gps as above and have followed the wiki and youtube links to get the GPS working. I have tried switching tx/rx wiring, different baud rates, different ports. Totally stumped!

The GPS tab has no activity at all, the light on the GPS unit flashing blue to indicate there is some signal. 

CLI Dump:
# version
# INAV/MATEKF405 2.0.0 Aug 20 2018 / 19:17:50 (dbdd1656a)

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature CURRENT_METER
[... 863 lines cut ...]
set roll_rate = 35
set pitch_rate = 9
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# 

[image]

giacomo892, 2018-09-20:
Try connecting the GPS module to ucenter using an USB adapter to check if it is outputting something. 
F405 + GPS is a common setup and it should work. I suppose you make some mistakes in the wiring/configuration or the GPS unit is defective / dead.

bookworm1707, 2018-09-20:
Thank you, I am investigating ucentre and what I need for it. This is the second GPS module. It has the power light come on and the signal light also comes on to say its receiving something. (The first unit didn't have any lights so I had no idea if it was working or not.

closed 2018-09-20 by bookworm1707