iNavFlight/inav #3876: Impossible to choose right MPU9250/9255 compass alignmnet

vasimv, 2018-09-20:
I've connected 10-dof board with MPU9255 by I2C to modified FLIP32 AIO board (replaced MCU with STM32F303CCT6) and found that can't make MPU9255's compass working right. Every possible orientation i choose causes either permanent or temporal (gyro changes heading at first but then filter slowly moves it in other direction) wrong heading.

May be i'm wrong but from MPU9255's datasheet it looks like you just can't make the MPU9250/5 compass working because gyro and compass parts aligned in weird way inside the chip. You need to invert one of axis'es without flipping whole to make it working.

In my case, i've done that by changing AK8963 (because cannot use MPU9255's in I2C mode) code compass_ak8693.c in that way:

```
#ifdef MAG_AK8963N
    magData[Y] = parseMag(buf + 0, mag->magGain[X]);
    magData[X] = parseMag(buf + 2, mag->magGain[Y]);
    magData[Z] = -parseMag(buf + 4, mag->magGain[Z]);
#else
    magData[X] = parseMag(buf + 0, mag->magGain[X]);
    magData[Y] = parseMag(buf + 2, mag->magGain[Y]);
    magData[Z] = parseMag(buf + 4, mag->magGain[Z]);
#endif
```

vasimv, 2018-09-20:
You can close it now. I've downloaded latest sources of 2.0.0 and found this is already fixed. Sorry, my mistake.

shellixyz, 2018-11-19:
Fixed in 2.0.0, closing

closed 2018-11-19 by shellixyz