iNavFlight/inav #3877: Hello everybody!

PanayiotaCy, 2018-09-20:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)
* [IRC channel](irc://irc.freenode.net/#inavflight)
* [Slack channel](https://publicslack.com/slacks/inavflight/invites/new)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

rb1205, 2018-09-20:
RX_PARALLEL_PWM works only if you're not using a serial RX provider. That is, it works only if you have one wire for each channel going from your RX to your FC. This kind of setup is outdated and hard to come by nowadays and, considering that your FC keeps disabling it, it probably mean that it's not what you're using.

If that's the case, then a [simple RC filter](http://www.etc.cmu.edu/projects/flux/wp-content/uploads/2013/02/scales_PWM-to-analog.png) can turn your PWM signal to a voltage signal that you can feed to the ADC RSSI input.

PanayiotaCy, 2018-09-21:
Thank you rb1205,

I managed to read an analog signal from the RSSI input, however it does not solve my problem. I am trying to trigger the RTH using an external pulse. Since the PWM input is not available on this FC i can use the analog RSSI but i would like to trigger the RTH using the RSSI input. In the modes menu of INAV at RTH i am only capable of selecting a Channel from Ch5 to Ch18. There is no RSSI or other ADC input option. Is there a way to do this? 

If not, is there a way to map the Analog RSSI value to one of the available Channels? For example, to assign a spare channel to linearly map the analog voltage range to its PWM range (1-2ms) instead of using the default SBUS input. The rest of the channels should remain on SBUS.

rb1205, 2018-09-21:
If you're using SBUS or any other serial RX provider then no, there's no way to mix PWM channels in.

Besides, using RSSI as an actual AUX channel to trigger RTH would be far from ideal. You wouldn't have all the checks and safeties you usually get with a proper failsafe RTH, like a cooldown period and stick disable. Given the fluttering nature of RSSI signals, you'd probably get an hard time when you're approaching the limit of your RC link. Also, a regular RTH is different from a failsafe RTH as in the former the flight and AUX channels are still valid and used, while in the latter are ignored as they should be.

So, my advice is to stick with an usual failsafe and just use the RSSI signal for telemetry information.

PanayiotaCy, 2018-09-21:
Thanks for your answer. I think you did not fully understand my second question. I am not trying to use the actual RSSI signal to trigger RTH. I have a signal coming from an external circuitry which is robust and i designed it to output 2 separate values PWM values. I then followed your suggestion to convert the PWM signal to Analog using the RC filter. Now i can read this signal in INAV RSSI and it reads 2 discrete values (20% and 80%). Now i want to trigger RTH using these discrete values or a short range around them to be more secure. I believe that the problem now is software dependent. If INAV could take the RSSI value coming in from the relevant ADC channel then the RTH should be triggered as desired.

rb1205, 2018-09-21:
I see now. Well, since you already have external circuitery why don't usa a BJT to short the SBUS signal to the ground, thus forcing the FC to failsafe and initiate a RTH?

closed 2018-12-04 by stale[bot]