iNavFlight/inav #3878: Led strip color issue on Omnibus F4 Pro V2

Vissie26, 2018-09-20:
When I connect a WS2812B Led to M5 on my omnibusF4 V2 (it's a clone) the color red doesn't display correctly (mixes a tiny bit of blue) but when I select Gps status it displays red correctly. 

I would expect the led to be full red instead of red with a bit of blue as well.

Could be a timer issue or a signal voltage issue.

When I flash the board to Betaflight it works perfectly

---
- FC Board name and vendor: Omnibus F4 pro V2 vendor: hobbyking
- INAV version string: INAV/OMNIBUSF4PRO_LEDSTRIPM5 2.0.0 Aug 20 2018 / 19:11:29 (dbdd1656a)

simtcrom, 2018-09-22:
Are you able to make the turn indicators work? 
For me left turn not working. Only right. And right even blinks when moving stick down too?

Vissie26, 2018-09-22:
For me the turn indicators don't work either. I have the same problem as you described. When I move the stick right the right leds blink and when I pull the stick down they blink too. Up and left don't work.

teckel12, 2018-09-22:
I've never got the turn LEDs to work perfectly on any version of INAV. I settled for just white. With flashing for battery low warning.

Vissie26, 2018-09-22:
How was the color red for you on the leds? mine (ws2812B from banggood and a local source) have a bit of blue simultaneously and I can't get rid of it. (If I flash the flightcontroller to betaflight it works fine)

Tygrys-1, 2018-09-25:
On the Omnibus Pro v2 (the OMNIBUSF4PRO target) the LED strip signal is available on the j4 connector CS signal. (the first from the right looking from the front at the connector). It might be more convinient to take it from there. There is also a GND and a 3V3 VCC on the same j4 connector.

Vissie26, 2018-09-26:
If I understand correctly. The pin you describe only works on an original Omnibus F4 Pro? I had to use the OMNIBUSF4PRO_LEDSTRIPM5 target to get the led strip working (clone omnibus messed up on wiring it correctly) Where would I connect the 3v3 VCC to? The datasheet of the WS2812 suggests a vcc of 3.5V to 5.3V VCC

Tygrys-1, 2018-09-26:
There is an error in Omnibus PRO V2 wiring and the 3 pin J1 Led Strip socket is unusable for the led strip. So there is a reason for the OMNIBUSF4PRO_LEDSTRIPM5 target. But the right Led Strip signal is present on that board on the J4 connector. So you can take it from there and use the standard target.
On V3 this situation is corrected and the right signal for the led strip is moved from J4 to J1.

The led strip will work on 3V3, just will be dimmer a little then on 5V. You can take the 5V from anywhere from the board and power the strip, but mind that the overall 5V supply is limited to 1A due to the diode which will overheat if you take to much current. If you plan to use a large strip it will be better to use the external BEC for it.