iNavFlight/inav #3880: Bidirectional Mavlink

pilotnbr1, 2018-09-22:
Does INav support mavlink channel override? 

Currently I am flying an omnibus F3 via a raspberry pi zero running an Ez-Wifibroadcast image. I have no traditional receiver onboard- the pi simply forwards sumd control to a uart on the F3. It would be better if I could use mavlink instead of sumd as I would have more of a bidirectional link.

pilotnbr1, 2018-11-24:
Not everyone at once now

digitalentity, 2018-11-24:
> Does INav support mavlink channel override? 

No

> To what degree is mavlink implemented?

Telemetry reporting. Waypoint upload (to some degree)

closed 2018-11-24 by digitalentity

shellixyz, 2018-11-24:
@pilotnbr1 Why not use MSP ? Native bidir proto

digitalentity, 2018-11-25:
@pilotnbr1 MSP would require work on your side, MAVLink will require work on INAV side. Seems nobody is interested in extending MAVLink beyond minimal implementation, you have two choices:

 1. Either implement MSP support for Ez-Wifibroadcast;
 2. Or implement MAVLink improvements for INAV.

pilotnbr1, 2018-11-25:
Yes... I like INav but may go to ardupilot (with my compatible INAV hardware) because of better mavlink support. Like it or not MAVLink seems to be the de facto standard and more widely supported with GCS software