iNavFlight/inav #3881: Problem with Inav 2 and omnibus F4 v3 PRO not ARMED

lumiror, 2018-09-23:
I have a problem with Inav 2 and Omnibus F4 v3 because not armed. I suspect an engine driver problem because I can not connect to BLHeliSuite and control and setup motors. In version 1.91 everything works correctly. Do you know where to be a problem?

c4xp, 2018-09-24:
I might have the same problem. If the sensor is ICM20608 it is confirmed, if not I will open another issue.
[image]
[image]

c4xp, 2018-09-24:
P.S. The board is this: https://www.banggood.com/HAKRC-Omnibus-F4-V3-Flight-Controller-OSD-w-5V-3A-Power-Distribution-Board-for-RC-Drone-FPV-Racing-p-1284817.html

lumiror, 2018-09-24:
Not this one https://www.aliexpress.com/item/FLIP-32-F4-OMNIBUS-V3-PRO-Flight-Controller-Board-w-Baro-built-in-OSD-For-RC/32827160882.html?spm=2114.search0604.3.28.1fc84f8fKNrvGf&ws_ab_test=searchweb0_0,searchweb201602_5_10065_10068_10059_10884_10887_10696_100031_10084_10083_10103_10618_10307_10820_10821_10303,searchweb201603_45,ppcSwitch_2&algo_expid=248675c1-6289-4347-aae1-8eeb151882f4-4&algo_pvid=248675c1-6289-4347-aae1-8eeb151882f4&priceBeautifyAB=0

c4xp, 2018-09-24:
Can you post the 'status' CLI text here. Like I have in the picture above.. It will help the devs.

lumiror, 2018-09-24:
I'll send, but after the afternoon

lumiror, 2018-09-24:
Here it is
[image]

lumiror, 2018-09-24:
I already know where the problem is, I have not set up a new mixer function. Now it's working

closed 2018-09-24 by lumiror