iNavFlight/inav #3882: Inav custom Version with UART1 MSP tuned on.

simtcrom, 2018-09-23:
Hello,

I broke micro usb port on omnibus F4 pro v2 and I am trying to regain ability to connect to computer back using ftdi adapter.
I guess for this to work, MSP should be turned on any UART.
Is there a way I can do a custom build where MSP is turned on default on UART1?

Thanks.

datamann, 2018-09-23:
Take a look at this video
https://youtu.be/5VmhoyL6Z7Q

simtcrom, 2018-09-23:
> Take a look at this video
> https://youtu.be/5VmhoyL6Z7Q

I had seen that video. And it is betaflight.
I want INAV.

lumiror, 2018-09-23:
And why do not you fix it?
[image]

simtcrom, 2018-09-23:
> And why do not you fix it?
> [image]

I attempted to. But they keep coming closer and getting short. Now the pads are gone too. So I don't know if there is a way to solder it.

lumiror, 2018-09-23:
Are these 4 pads damaged?
[image]

simtcrom, 2018-09-23:
> Are these 4 pads damaged?
> [image]

Those 4 pads you have highlighted are not damaged. By the way mine is omnibus F4 pro v2 (clone) Target - OMNIBUSF4PRO_LEDSTRIPM5

[image]

lumiror, 2018-09-23:
so that soldering

lumiror, 2018-09-23:
I do not remember it, look at the internet. You need only  4 wires, the fifth is grounding (do not need to connect)

simtcrom, 2018-09-23:
> I do not remember it, look at the internet. You need only 4 wires, the fifth is grounding (do not need to connect)

Is it safe to assume first and last pads are 5v and gnd respectively? if so I could try other two wires straight and if not work swap around?

[image]

lumiror, 2018-09-23:
So
[image]

simtcrom, 2018-09-24:
> So
> [image]

I wired it like this way and tried connecting. FC got powered up, but computer says "Unknown USB device". Before I broke USB, it was getting detected correctly.

simtcrom, 2018-09-24:
I had to invert D- and D+ and that fixed issue.
Still trying to make the wires stick to the pads.

lumiror, 2018-09-24:
try to soldering here
[image]

simtcrom, 2018-09-24:
> try to soldering here
> [image]

I did the original way you suggested and out hot glue on it. Its working now. 
If it ever come off, I shall attempt this way. Thanks a ton, you saved my work.

simtcrom, 2018-09-24:
Used the pads lumiror suggested and all good now.

closed 2018-09-24 by simtcrom