iNavFlight/inav #3883: Fixed Wing RTH odd behavior

flavioansovini, 2018-09-23:
Hello, I write to ask for help on a problem related to the rth that I cannot solve by myself.
My setup is composed as follows:
plane: Easystar Multiplex
fc: Omnibus F4 V3
tx: taranis + crossfire
fw: INAV 2.0.0
The plane fly pretty good in manual (how i usually fly) or air mode, at the moment i'm testing the rth and i found out that sometimes has a strange behaviour.
For example with 10knot from W, i was flying to S (180° from home) when i triggered the rth. If the plane turns to E it makes a large turn and finally go to home, but if it turns to W (upwind) it makes a very tight turn and suddenly enter in a descending screw (i had to save it in manual). 
It seems to be a piff config issue.. but i've also notice that even when everything seems to be good and the plane, already aligned, it is going back home, sometimes change the direction by himself and go again in the descending screw.   
Before writing i've already triyed several piff config, starting from the default values, 
https://github.com/iNavFlight/inav/wiki/Fixed-wing-guide
and following the piff calibration guide
https://github.com/iNavFlight/inav/wiki/Tune-INAV-PIFF-controller-for-fixedwing
the response of the plane changes of course, but the end is always the same!
To better understand what i mean i have a video with several rth test (wind abt 10 knot from S)
[https://youtu.be/oJvuUuETPvE](https://youtu.be/oJvuUuETPvE)
you can jump at this points
0:00 - 00:14 - good rth
0:42 - 01:05 - good rth
1:48 - 02:02 - good rth
2:56 - 03:10 - bad rth, falls in a screw suddenly
3:42 - 04:00 - odd behaviour, seems to be good but at the end.. surprise!
Would be nice to have some opinions from you, because now i'm short of idea.
Sorry for the long post,
Thanks in advance, Flavio

Tygrys-1, 2018-09-24:
I have some Spiral of Death behavior on S800 Sky Shadow when flying in angle or horizon mode and turning. The plane cannot resolve the problem itself in angle mode, if not taken out of the SOD it falls to the ground. I'm still learning how to react in that situation, usually releasing a sticks and switching to angle mode levels the plane, but not in this case.
I think that on default PIDs (iNav 1.9.1) it behaves better, but after the autotune procedure it is more prone to this behavior now. 
Just my thoughts, I haven't experimented with this, I have no blackbox on the controller.
I'am surprised that in your case it happened in automatic mode, as it makes things dangerous for the RTH or missions.

flavioansovini, 2018-09-24:
Thanks for your comment Tygrys-1, sure it is dangerous, rth should bring it back in a fail condition.. I suppose that even with a non perfect piff setup, the plane should fly somehow in a wrong path.. but falling in that way seems to be too strange to me. Consider that your S800 is more critical then my easystar, smaller and faster..
Before INAV 2.0 i flown with INAV1.9 and i didn't realize this behaviuor, but i did only some short rth tests (with no problem)

flavioansovini, 2018-10-05:
Thanks b14ckyy, i'll take a look at your posts.
Fla

b14ckyy, 2018-10-05:
Btw. I have another plane (Flite Test Bushwacker) with an SPRacing F3 deluxe and iNAV 1.9.1 and that works fine in RTH. Not sure right now if I already updated it to 2.0 during the last weeks because I had no time and weather to fly it. I could check that later. I also had a cheap foamy cessna with SPRacing F3 deluxe (but the plane is disassembled because it was crap in many ways) and thgat was flown with 1.9.1. Also never had any RTH issues here. Will see if I find some time to fly the Bushwacker again if there is 2.0 to do some more tests. 
At this point, all reports of RTH failures seemed to be WIngs and not planes. Possibly some mixer issues because of the missung rudder authority? But that doesn't expolain the sudden unlimited altitude raise I observed on my AR Wing.

flavioansovini, 2018-10-08:
I came back to INAV 2.0 and seems that now is ok, at least it doesn't falls down suddenly! 
I think that the problem was the tendence of the plane to pitch up on throttle up (.. and stall), i've decreased the cruise throttle and stabilization on rth is ok, but it is not precise on the heading.
Should i have to increase nav_fw_pos_xy_p and nav_fw_pos_xy_i?
Thanks Flavio

sniperxnl, 2018-11-14:
I just got the same problem
I hit rth, comes back, bur after a minut it get a spiral of dead

flavioansovini, 2018-11-14:
hello sniperxnl, you may try to lower the p values, seems that the piff controller can't converge if the p is too high and d ( it's ff really) is too low.

sniperxnl, 2018-11-14:
@flavioansovini  
I have done auto tuning before, and have used these pids for my Capriniha 2
See dump 
[F35 inav.txt](https://github.com/iNavFlight/inav/files/2581035/F35.inav.txt)

Video what happends: 
after 30 seconds: https://www.youtube.com/watch?v=7AXWIbvuIr8&feature=youtu.be

Feldsalat, 2018-11-14:
At least this is an old issue: 
[image]

Could that also be a cause in this case?

sniperxnl, 2018-11-14:
Old issue?
Using inav 2.0

flavioansovini, 2018-11-14:
Hi sniperxnl, i think that you have some problem related to the attitude estimation..  maybe calibration or wrong acc,gyro setup.. if i look at 0:37 the plane is looking at left but the FC thinks that it's angled to the right more and more.. it won't compensate if the attitude is wrong...

Tygrys-1, 2018-11-14:
> @flavioansovini
> I have done auto tuning before, and have used these pids for my Capriniha 2

Autotune seems to favour higgh FF with lower P-gain, which might give less authority to the FC, as said in [https://github.com/iNavFlight/inav/issues/3887#issuecomment-424814451](https://github.com/iNavFlight/inav/issues/3887#issuecomment-424814451)

sniperxnl, 2018-11-14:
Weird thing is, i am using a F35 and my friend have the same problem with a omnibus. 

Calibrated everything.

flavioansovini, 2018-11-14:
i bet that the problem is the mag... try to test the attitude with and without the internal mag enabled using the INAV gui... probably without will work mutch better and faster. i have the f35 too and i solved using the naza gps+mag (because the f35 doesn't even have the i2c connector available for the mag, but the naza gps send the mag data through th serial port with the gps data).

sniperxnl, 2018-11-14:
@flavioansovini 
Why my friend with a m8n gps with compass have the same problem then?
F35 i2c is avalible on the pitotube port btw

sniperxnl, 2018-11-14:
> > @flavioansovini
> > I have done auto tuning before, and have used these pids for my Capriniha 2
> 
> Autotune seems to favour higgh FF with lower P-gain, which might give less authority to the FC, as said in [#3887 (comment)](https://github.com/iNavFlight/inav/issues/3887#issuecomment-424814451)

So you advise to increase this?
[image]

flavioansovini, 2018-11-14:
isn't the pitot tube analog on the f35?

flavioansovini, 2018-11-14:
after seen your video i think that your problem isn't related to the piff values but the attitude estimation.

Tygrys-1, 2018-11-14:
I suggest to try the PID defaults, as they might be better then autotuned ones. I found my S800 wing more prone to SOD after autotune. Just an impression, not hard testing.

shellixyz, 2018-11-14:
@sniperxnl Your issue is with the attitude estimation. The artificial horizon goes completely crazy in your video. If the attitude is wrong the plane will turn weirdly but even worse the throttle management will be completely wrong. In your video you can see that the horizon is much higher than it should be. So the plane is trying to pitch up without enough throttle and stalls.

closed 2019-01-28 by stale[bot]