iNavFlight/inav #3884: ICM20608 does not pass pre-arm check: Run-time calibration.

c4xp, 2018-09-24:
Plane does not pass pre-arm cheks. If the sensor is ICM20608 it is confirmed.
- The board is this: https://www.banggood.com/HAKRC-Omnibus-F4-V3-Flight-Controller-OSD-w-5V-3A-Power-Distribution-Board-for-RC-Drone-FPV-Racing-p-1284817.html
- Inav version 2.0 final
- I switched in the meawhile to another FC so I can provide any requested feedback to help any development with the board on my desk.
[image]
[image]
- The only Gyro that works is MPU6500 in the configuration

Tygrys-1, 2018-09-24:
Have you calibrated the sensors? It says that it is not calibrated.

c4xp, 2018-09-24:
I've done Accelerometer Calibration 6times. As you can see in the image "Accelerometer calibrated".
The page https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons says it is related to "The pre-arm sensor calibration has not completed".
That calibration is in fact the gyro sensor not the accelerometer.

Tygrys-1, 2018-09-24:
Yes, you are right. I missed that.
I once have the arming problem when the model was laying on the car with the working motor. The vibrations were to high. Your gyro is very sensitive. Maybe there is some kind of vibration or interference to the signal. Can you look at the sensors tab and see what gyro shows? For calibration to succeed it should be still for some time.

c4xp, 2018-09-24:
ICM20608 is indeed more sensitive than MPU6500. The values keep fluctuating. But the issue is with inav v2. Betaflight works, which let me to believe is from current software implementation.

digitalentity, 2018-09-24:
It may as well be barometer on your board. Also to keep in mind - INAV has a lot more sanity checks whereas Betaflight only cares about gyro. In general defaults are good enough to work with ICM20608 (verified). If it consistently fail - it's likely a hardware issue.

c4xp, 2018-09-24:
[image]

c4xp, 2018-09-24:
I just did
```set small_angle = 180```
same issue :(

c4xp, 2018-09-24:
> 
> 
> It may as well be barometer on your board. Also to keep in mind - INAV has a lot more sanity checks whereas Betaflight only cares about gyro. In general defaults are good enough to work with ICM20608 (verified). If it consistently fail - it's likely a hardware issue.

Maybe it's from the target I picked?
```# version
# INAV/OMNIBUSF4V3 2.0.0 Aug 20 2018 / 19:10:22 (dbdd1656a)```

Tygrys-1, 2018-09-24:
Disable baro in the configurator and check.

Flash iNav 1.9.1 and check

The board was released for iNav 1.9.0 according to the product page, check with this if all else fail.

You may also disconnect and reconnect the gyro ribbon cable - as a part of magic that sometimes work. ;-)

c4xp, 2018-09-24:
Disabled baro. Fail.

Tygrys-1, 2018-09-24:
IMHO It is moving too much on the screen you provided.

c4xp, 2018-09-24:
Flashed 1.9.1. Fail. Same drift but with the old drone 3d image instead of that big ugly box :)

Tygrys-1, 2018-09-24:
The drone is the default if you have quadcopter configured. For airplane it is a box.
I susspect hardware problem. :-(

c4xp, 2018-09-24:
Disconnected-reconnected cable. Fail.
Hmm I dont think it's a hardware issue because changing the orientation in space moves the 3d model in the correct directions.

digitalentity, 2018-09-24:
Relative motion in space is calculated by gyroscope. Most likely accelerometer is reading garbage. This FC is unsafe to fly anyway.

Tygrys-1, 2018-09-24:
Not that kind.
Rather high noise from somewhere. Maybe in the chip itself. If laying flat it should not move the model. /Unless you live in the tectonic movement area and haven't even pay attention to that ;-)/

c4xp, 2018-09-24:
Ok :( Can't really dissprove your point. I will buy another. I really like this FC because the layout and the price.

Tygrys-1, 2018-09-24:
Unless you do not buy directly from the Airbot or some known company it is all the Chinesee crap. And there is no waranty at all for that, on my Omnibus Pro v2 I found SD card shield touching one resistor and preventind a beeper to work. Or one diode mounted in the reverse direction. And I heard simmilar stories here and there. May you try to change the gyro chip only?

c4xp, 2018-09-24:
Interesting proposition. I will give it some thought and research. And you are right about chineese crap :))
I did a flash with betaflight v3.5.1 and shows the same drift, although can be armed.

```
# status
System Uptime: 24 seconds
Current Time: 2018-09-24T19:13:34.033+00:00
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=168MHz, Vref=3.29V, Core temp=34degC, GYRO=ICM20608G, ACC=ICM20608G, BARO=BMP280
SD card: None inserted
Stack size: 2048, Stack address: 0x1000fff0
I2C Errors: 0, config size: 2579, max available config: 16384
CPU:7%, cycle time: 129, GYRO rate: 7751, RX rate: 33, System rate: 9
Arming disable flags: RXLOSS CLI MSP
```

digitalentity, 2018-09-24:
@c4xp which explains alot. Betaflight doesn't care about accelerometer.

Tygrys-1, 2018-09-24:
Well, betaflight will let you crash on your own :-) As digitalentity said, iNav is more demanding in case of the safety.

c4xp, 2018-09-24:
That's why I'm here :)) I like where this is going and would like to lend a hand.

Tygrys-1, 2018-09-24:
BTW, if you contact the BG customer service they may compensate it to you somehow, or even return the cost. Just send them the evidence and negotiate on their first offer :-)

c4xp, 2018-09-24:
BTW Can you guys recommend a cheap F4 board for beginners and kids with 6 servo outputs, barometer, current sensor and maybe* a builtin Pdb :D Yeah I know, I'm wishing..
Until now this is gets pretty damn close:
https://www.banggood.com/Original-Airbot-Omnibus-F4-Pro-V3-Flight-controller-SD-5V-3A-BEC-OSD-Current-Sensor-LC-Filter-p-1319177.html

digitalentity, 2018-09-24:
@c4xp that look pretty close to original Airbot OmnibusF4 Pro - a well known and quite popular target

digitalentity, 2018-09-24:
You can also check out https://www.banggood.com/Matek-Systems-BetaFlight-F405-AIO-Flight-Controller-Built-in-PDB-5V-2A-9V-2A-Dual-BEC-for-RC-FPV-Racing-Drone-p-1165338.html

Tygrys-1, 2018-09-24:
Or you may have 2 of this in the price of the other one https://www.aliexpress.com/store/product/F4-PRO-V2-Flight-Controller-Built-in-OSD-BEC-Drone-for-Betaflight-for-OMNIBUS-Low-Noise/2128265_32891137083.html

c4xp, 2018-09-24:
I dunno :D aliexpress is the m**r of c**p :D but that price..

Tygrys-1, 2018-09-24:
I got two of them recently and both are working ok. Even all diodes are in good direction :-)
I do prefer the V2 version because thay have more pads for softserial's and so, which later builds lacks.
They have also the less sensitive gyro. Overall a good board. Only the LED strip is on the other socket then described and if you use battery voltage for the camera from the onboard LC filter you may have something called CLUB 24 problem. I use external BEC for the FPV so I don't care.

For better quality I'll go for the Matek F405. It has also the Wing version, more expensive, or you may add the PDB for the wing or quad. Matek is a super resource-wise board, you can have lot of uarts, motors, servos and whatever. Imagine - on every CPU output Matek adds a 200 ohm resistor for the protection. But not that cheap.

c4xp, 2018-09-26:
Banggood was kind enough to supply another board. Let's test this :)
Btw thank you Banggood ;)

lumiror, 2018-09-26:
> Nebo můžete mít 2 z toho v ceně druhého [https://www.aliexpress.com/store/product/F4-PRO-V2-Flight-Controller-Built-in-OSD-BEC-Drone-for- Betaflight-pro-OMNIBUS-Low-Noise / 2128265_32891137083.html](https://www.aliexpress.com/store/product/F4-PRO-V2-Flight-Controller-Built-in-OSD-BEC-Drone-for-Betaflight-for-OMNIBUS-Low-Noise/2128265_32891137083.html)

yes half price but not original:
http://www.readytoflyquads.com/flip-32-f4-omnibus-v2-pro

c4xp, 2018-10-22:
I just received and tested the exact same FC. It is working correctly so this was indeed a hardware fault.
This ticket can be closed :)
Thanks everyone..

closed 2018-10-22 by c4xp