iNavFlight/inav #3887: Wing tip stall spin recovery

danarrib, 2018-09-26:
## Current Behavior
When in ANGLE mode, and insufficient thrust to maintain level flight, Fixed Wing aircraft will stall. On flying wings, the stall usually starts on the wing tips as it produces less lift than the wing roots. Because of that, when stalling, a Flying wing tend to do a hard dive and start spin on roll axis.

ANGLE mode will try to level airplane again by throwing both control surfaces up. But this cause the wing to keep spinning and don't make it level again... So it'll dive spinning to the ground.

## Desired Behavior
ANGLE should be able to detect this situation, and fix or make it easier to pilot fix the situation.

## Suggested Solution
ANGLE mode should have a sanity check. If aircraft is diving and spinning for some time (like 0.5 sec) and ANGLE is unable to control aircraft again, then INAV could alternate to ACRO mode, so control surfaces can return to the neutral position (or respect the input from the pilot). Then, when aircraft stops spinning, ANGLE mode can be re-activated again.

## Who does this impact? Who is this for?
Airplane users (Specially Flying Wings), Beginners that don't have enough experience to be fast to detect the situation and remember to turn ANGLE mode off.

## Additional context
Video that shows the problem (Aircraft is a Z84, Inav 1.9.1)
https://youtu.be/osya99vOap4

Tygrys-1, 2018-09-26:
I noticed this behavior on S800 INav 1.9.1. But I don't think the acro will help here, as the acro does not autolevel the plane and will not help the situation. I found that cutting out the throttle and aplying full up elevator in Angle mode stops the spin, but I have to do more tests with this or other solution. There is a very short time you can react before the crash.

My observation is that on default PID settings it is less prone to start the spiral then on the tuned ones where P value are much reduced.

What worse, there is a report here that this situation occured in RTH mode, when pilot has no authority on the surfaces. So it has to be handled by INav itself, not relaying on pilot's input.

Tygrys-1, 2018-09-26:
#3883

danarrib, 2018-09-26:
> But I don't think the acro will help here, as the acro does not autolevel the plane and will not help the situation. 

Acro in fact doesn't level the aircraft. But it makes aircraft stop spinning. Once aircraft is not spinning (just diving), ANGLE mode will be able will be able to level the aircraft again.

digitalentity, 2018-09-26:
First thing to recover from stall is to gain airspeed. Stall (loss of control) occurs because elevon surface is not enough to keep the aircraft level any more. Neither acro nor angle will be able to recover.

danarrib, 2018-09-26:
> First thing to recover from stall is to gain airspeed. Stall (loss of control) occurs because elevon surface is not enough to keep the aircraft level any more. Neither acro nor angle will be able to recover.

Aircraft is diving, so airspeed is increasing. At some point it'll be enough to allow aircraft to fly again.

Tygrys-1, 2018-09-26:
The problem with the stall is another one. The idea is to exit the Spiral of Death. And I think it is how the iNav in Angle mode handles the abnormal rotation. At the first time I tried to switch to the angle mode when it happened, as it always (or untill this time) leveled the plane. But it didn't helped and the plane crashed. And I think it occured after the PID autotune, where P gain was reduced from the default of 15 to 3 on roll and 8 on pitch. I think this may be the problem here.

digitalentity, 2018-09-26:
Indeed. PIFF controller with emphasis on P is actively counteracting rotation (this, however, stresses servos much more).

Tune with emphasis on FF-term will deflect elevons mostly according to angle error and will not be able to stop the abnormal rotation. This is the tune that minimizes the wear on servos. Autotune tunes this way.

Tygrys-1, 2018-09-26:
I think it is good overall, but it should handle SOD somehow. In BF there is a mechanism to temporary change some PID on i.e. quick throttle movement. Maybe the iNav could handle SOD by temporary increasing the P-gain? At least in angle/automation modes.

OlivierC-FR, 2018-09-26:
I noticed this behaviour, my Z84 did that when the COG was a bit off, as danarrib said the Angle mode is then actively maintaining the spiral by trying to keep the plane horizontal. That's an easy task to get of such event for a flight controller, it's already quite difficult for a decent pilot, must reduce throttle, dive the plane hard, gain speed and then pull up, if the altitude is not enough there's simply not enough time to do this. I honestly don't think a FC could do it in a safe and reliable way :(

danarrib, 2018-09-26:
What I'm suggesting is just to INAV disable ANGLE mode for a while when it detects that it's not succeeding to keep aircraft level... Aircraft is not flying anyway, so in worst case, you'll crash the aircraft anyway.

ANGLE mode is making difficult for pilot to regain control over the aircraft, so in any case, it's better to disable it and let the laws of physics and the pilot skills try to correct this situation instead of keep trying to level with a method that will not work.

teckel12, 2018-09-27:
@danarrib Or the pilot could just change into horizon mode if angle mode wasn't cutting it. I've done that before on an underpowered quad on a windy day.

digitalentity, 2018-09-27:
The problem can not occur in automated modes if your cruise throttle is set correctly to maintain sufficient airspeed with a safe margin. All stalls I have seen were caused by insufficient airspeed which in turn was caused by insufficient throttle.

danarrib, 2018-09-27:
Yes... I'm talking about stall in user-operated ANGLE mode, not automated modes. As @digitalentity said, if aircraft is stalling in automated modes, you probably need to increase the cruising throttle to a higher value.

Pairan, 2018-09-28:
https://youtu.be/1bcwa-rVwrQ

Dronek, 2018-09-30:
> As @digitalentity said, if aircraft is stalling in automated modes, you probably need to increase the cruising throttle to a higher value.

Do the same with your throttle stick in non-automated modes and you will be all good ;)
Let's not encourage clueless flying. 

If it's about increasing safety then stall prevention would be a more practical way to go than stall recovery. This is for example successfully realized in the Parrot Disco where it's impossible to apply too less throttle in flight and the minimum throttle limit is dynamically adjusted based on airspeed (lower when diving / higher when climbing, etc.).

giacomo892, 2018-10-01:
@Dronek Note that a product like the Disco have been heavily engineered and the flight envolope is well known by the FC. Also it has airspeed sensor so enforcing the correct airspeed for the airframe will make sure it never stalls. 

Regarding a sorta of stall recovery not sure if some automated wild guess could make a plane recover. I mean, I'm not sure if disabling momentarily ANGLE mode forcing the plane to dive a bit in AIRMODE could save it reliably before switching back to ANGLE as soon the rotation stopped.

Dronek, 2018-10-01:
> @Dronek Note that a product like the Disco have been heavily engineered and the flight envolope is well known by the FC. Also it has airspeed sensor so enforcing the correct airspeed for the airframe will make sure it never stalls.

And I think it's more reasonable to aim for a similar flight support (for pilots who need it) in iNav instead of stall recovery. In both cases the user will have to deal with configuration of parameters but stall prevention will have a wider range of use-cases.

danarrib, 2018-10-01:
> Regarding a sorta of stall recovery not sure if some automated wild guess could make a plane recover. I mean, I'm not sure if disabling momentarily ANGLE mode forcing the plane to dive a bit in AIRMODE could save it reliably before switching back to ANGLE as soon the rotation stopped.

I'll test it... I'll fly > 500m altitude, then force a wing tip stall in ANGLE mode (just cut throttle and wait for the stall), then I'll just let it dive spinning for a while, then I'll deactivate ANGLE mode. When it stops spinning, I'll activate ANGLE mode again. I will not touch any sticks in the process.

I'll post the DVR video as soon as I can.

Also, I can't say for sure, but I'm pretty sure that Parrot did some kind of stall recovery process as a "second layer of security". It's obvious that prevent stall is always the best way to go... But some wind shear can cause the stall even with a correct minimum thrust and let this possibility unassisted would be a problem for them.

closed 2018-12-14 by stale[bot]