iNavFlight/inav #3889: Mass Storage Device (MSC) Support

ZinhoRacing, 2018-09-27:
## Current Behavior
Not supported yet, available in betaflight 3.5

## Desired Behavior
Mass storage device support allows the micro SD card or the flash chip of your flight controller to be mounted as an external storage device on your PC. This allows for easy drag / drop download of log files.

## Suggested Solution
Merge from betaflight?
I don't have the skills to do it :(

## Who does this impact? Who is this for?
Every body.

teckel12, 2018-09-27:
I can't even flash my FC half the time without pressing the book key (which has worn off on a few FC).  I say we get flashing working 100% before we complicate things with MSC support (which is just as easy to pop the SD card out and stick in your computer anyway).

ZinhoRacing, 2018-09-27:
On my side, I can't acces the SD Slot without removing the FC from the plane ...
No big issue with flashing (Matek F405 wing / windows10 / standalone configurator).
But fair enought. I thought it would not impact flashing or so.

stronnag, 2018-09-27:
Duplicate of #1336

fiam, 2018-09-27:
Ack. Please, continue the discussion on #1336. Thanks!

closed 2018-09-27 by fiam