iNavFlight/inav #3890: [feature request] parameters for mission waypoints (time to stay, action, etc.), numbering waypoints

Pitichinaccio, 2018-09-27:
Hello,
I had the idea of adding some more parameters to the waypoints of a mission, like time to stay (until flying forward to next WP), actions (staying still, heading to certain direction, turning round with defined speed) and things like that.
There was also an other suggestion on the INAV group at facebook to add numbers to the waypoints and have them dislayed on the OSD.
Unfortunately my programming skills are to poor to do this on my own, but maybe it is an idea worth thinking about. Thanks!
Bernd

stronnag, 2018-09-28:
See the sadly unloved #154

StewLG, 2018-10-21:
[image]

I am working on your request to show waypoints and waypoint numbers on the OSD. Unfortunately I am running into performance problems due to the short amount of time allocated to OSD re-draw. 

https://github.com/StewLG/CraftServices

closed 2019-01-03 by stale[bot]