iNavFlight/inav #3891: Omnibus F3 (clones, 5 different boards) Fixed Wing NO Servo signal beyond pin 4

mickey2011, 2018-09-29:
Can someone check that there are control outputs on pins 5-8 on Fixed Wing Conventional aircraft. A Flying Wing using servo outputs 3 and 4 works OK but not for a Elevator, Aileron L, Ailron R, Rudder setup which would also use servo outputs 5 and 6.

This has been an issue from way back and still exists with the new Mixer setup in INAV vs 2.0.

My trials always use external 5 volt power supply for servos, always complete circular grounds/common.

INAV target = OMNIBUS

shellixyz, 2018-09-29:
There is no issue with the mixer. Outputs 5 and 6 are shared with UART3. To use them you need to disable UART3 and also if you want to use the PPM input remove the S.Bus jumper.

https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20OMNIBUS.md#usage-in-a-fixed-wing

mickey2011, 2018-09-29:
Thanks shellixyz for such a prompt answer, I've struggled with this for months, buying multiple Omnibus F3s

OK Outputs 5 and 6 are shared with UART3 and yes I have FRsky SBUS on UART3. BUT with the above setup I can drive 5 thru 8 servo's (such as a camera servo) via RC Channel 5-8 in the Mixer, they run OK from -100 to 0 but not so well from 0 to +100 and 8 is a bit different still.
Can you help me get around this issue, currently UART1 is GPS, UART 2 is Frsky Smartport Tele, UART 3 is Frsky SBUS RX.

I would ditch the Telemetry, but only if I have to.  Ah, how do I "disable UART 3" or do you mean ' just don't use it?

(Only GPS, not MAG)

Your help is appreciated. Thanks

shellixyz, 2018-09-29:
You mean you have servos connected to outputs 7-8 ? Outputs 7 and 8 are not configured as such by iNav. These are shared with the I²C bus. That's the issue with this FC and it is difficult to please everybody. I²C is needed to connect a mag for MR users and the mixer doesn't take into account if the outputs 5-6 are used for UART3 or not. If you want to use 7-8 you will need to build a custom firmware.

Alternatively what you can do if you want as many PWM outputs (8?) is to use a PCA9685 I²C IO expander.