iNavFlight/inav #3892: AK8975 not working over I2C

Zipdox, 2018-09-29:
So I'm trying to connect my GPS puck's compass over SDA and SCL but it isn't really working. The chip on the board is the AK8975 and so that's what I selected. But the magnetometer symbol was red after rebooting and the status dump:

```# status
System Uptime: 171 seconds
Current Time: 2018-09-29T17:34:11.479+00:00
Voltage: 11.62V (1S battery - OK)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=UNAVAILABLE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: None inserted
Stack size: 6144, Stack address: 0x10010000, Heap available: 884
I2C Errors: 0, config size: 4135, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = none
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1000, PID rate: 1000, RX rate: 49, System rate: 9
Arming disabled flags: NAV HWFAIL CLI```

When I measured SCL and SDA with my oscilloscope it showed SDA is constantly high and SCL is constantly low. Yes, it does receive enough power, I have tried 4.7k pullups and there's nothing on UART3.

Now interestingly when I set the magnetometer to HMC5883, there is _some_ traffic during startup but then it ends and SDA is constantly high and SCL is constantly low again. And sometimes, very rarely it works set to HMC5883 but then after a restart it's dead again. I'm not sure if this is a hardware or a software issue but since it behaves differently on different settings I think it's software. I'll try it with an arduino soon so'll report back later.

shellixyz, 2018-09-29:
Some pucks have a translator chip to fake the mag as an HMC5883 mag and they don't seem to work right. It is probably the case of yours. Some people have fixed the issue by removing the translator chip and bridge directly the SCL/SDA lines to the cable.

Zipdox, 2018-09-29:
[image]
I'm pretty sure it's the sanded one on the right.

Zipdox, 2018-09-29:
Do I just cut the traces then and attach the SDA ans SCL pins directly to the flight controller?

Zipdox, 2018-09-30:
I tried it and it doesn't work. Still nothing seems to be happening on SCL and SDA.

Zipdox, 2018-09-30:
It's only in the new version, confirmed. I flashed 1.7.3 and it worked right away! Victory!

closed 2018-11-14 by Zipdox