iNavFlight/inav #3893: Change the mode WP defaults to when there's no loaded mission

skorokithakis, 2018-09-29:
Currently, WP mode acts like full manual/acro when no mission is loaded. That is a dangerous default, as forgetting to load a mission will switch the craft to full manual instead of full auto. Having poshold/althold as a default would be much saner, as then the failure would be "my craft is unexpectedly holding its position" instead of "my craft is unexpectedly flying towards people".

simtcrom, 2018-10-01:
I launch the quad and enable pos hold and enable WP.
So any point WP is failed or cancelled, quad is in pos hold.

skorokithakis, 2018-10-01:
That's a good workaround, I didn't know WP worked in poshold. However, I think it would be best if this problem were fixed as well.

shellixyz, 2018-11-30:
I think that if the switch to the newly selected navigation flight mode is not possible (high PDOP, no WPs loaded for WP mode) the FC should leave the last flight mode enabled. For now it will switch to rate mode (ACRO).

stronnag, 2018-11-30:
My recollection is that is does leave prior last flight mode, at least for no WP loaded.

shellixyz, 2018-12-01:
@stronnag Not in my experience. I don't mix angle and navigation modes so when switching to a navigation mode if it doesn't engage it is like no mode have been enabled so it is back to acro.

@skorokithakis If it doesn't switch to WP it could be that the PDOP is too high or you don't have enough sats. Then it won't enable POSHOLD anyway so back to square one. At least if it doesn't change mode it shouldn't kill anybody because you can expect it.

skorokithakis, 2018-12-01:
@shellixyz In the case that PDOP is too high or I don't have enough sats, it's fine if it tries the next most stable mode, but if it does, there's no reason to not enable POSHOLD. Basically, the craft should always do the closest thing to what I meant, with consistency having a lower priority than predictability.

shellixyz, 2018-12-01:
POSHOLD needs GPS too. If WP doesn't engage because the position is not healthy it can't engage POSHOLD either. I guess it could engage ANGLE instead but I find it more logical for it to stay in the previous mode.

skorokithakis, 2018-12-01:
I know, I'm saying there's no reason to throw the baby out with the bathwater and say "because we can't POSHOLD with no GPS, we shouldn't POSHOLD when we have GPS either". I think the vast majority of people would expect that, when they trigger WP and it fails, the craft would stay in the mode that's closest to full-auto.

I think it makes much sense to say "You switched to WP intending to fly a mission, but since you forgot to load the mission I'm just going to stay in ACRO even though you took your hands off the sticks".

shellixyz, 2018-12-01:
The problem is that you should be able to know what to expect. If you don't know the position health you won't be able to know what mode it engages and that's even worse. If it can't engage a navigation mode and stays in the previous mode you at least know what mode you're in. If you are in angle mode, try to switch to WP and it doesn't engage it would stay in ANGLE. If you were in ACRO it would stay in ACRO. etc.

Also if it fails engaging a navigation mode the aircraft should beep angrily (for LOS) and also display a message on the OSD giving a warning and the reason.

skorokithakis, 2018-12-01:
> you won't be able to know what mode it engages and that's even worse

For me, "last mode" means "whatever the switch position before WP is", so it's not intuitive.

> you at least know what mode you're in

For the use case I'm talking about, I'm expecting the mission to start working and might not even realize I'm in another mode.

> if it fails engaging a navigation mode the aircraft should beep angrily

Unfortunately, with most non-stabilized modes, by the time I realize the aircraft is not flying itself, it's too late to recover, like what happened to me last time :/

shellixyz, 2018-12-01:
Non-stabilized modes ? There is only one non-stabilized mode that's manual mode (plane only).

What happened to you ?

shellixyz, 2018-12-01:
> For me, "last mode" means "whatever the switch position before WP is", so it's not intuitive.

That's an issue indeed if there are steps between the mode you switch from and the mode you want to switch to. A small delay could be added between flight mode switches. Then last mode will really be last mode not last position on the switches before the navigation flight mode.

skorokithakis, 2018-12-01:
I was in POSHOLD, tired to start the WP mission and the craft started moving erratically. I panicked and grabbed at the sticks but couldn't stabilize it (it was in ACRO and I was LOS). It was really bad because I was surrounded by buildings and not expecting anything to go wrong, but I don't remember if it ended up crashing in a tree or if I managed to enable POSHOLD just in time.

Needless to say, I'm trying to avoid this happening to anyone else.

stronnag, 2018-12-01:
> @stronnag Not in my experience. I don't mix angle and navigation modes so when switching to a navigation mode if it doesn't engage it is like no mode have been enabled so it is back to acro.

Which is exactly what I'm saying. If you're _de facto_ in ACRO, that's where it stays.

By design, I have WP on a separate switch to HOR or PH, and only switch to WP from these modes, so I deliberately enforce a safe mode hierarchy if I forget to upload WPs.

I also agree that there is no good 'fallback', because if you can't depend on e.g. PH if the reason for failing to engage WP is low sensor quality.

In my experience, there's no substitute for engaging brain and ensuring it's safe before one engages WP.

skorokithakis, 2018-12-01:
> By design, I have WP on a separate switch to HOR or PH, and only switch to WP from these modes, so I deliberately enforce a safe mode hierarchy if I forget to upload WPs.

Yet you seem against a default safety hierarchy for people whose hardware don't allow them to do that?

I don't understand the reasoning here. What you're proposing is making things less safe for every case:

Case | My suggestion | Your suggestion
---|---|---
WP fails, user thinks they're in ACRO | They are in POSHOLD, making it impossible to crash | Perception matches reailty
WP fails, user thinks they're in WP | They're in POSHOLD, making it impossible to crash | User is in ACRO, easy to crash
WP fails, POSHOLD fails, user thinks they're in WP | User is in ANGLE, making it hard to crash | User is in ACRO, easy to crash
WP fails, POSHOLD fails, user thinks they're in ACRO | User is in ANGLE, making it hard to crash | Perception matches reality

There is no case where your suggestion is safer than mine. When the user intends to do X, and X fails, the closest thing to X should happen. When switching on WP, the craft should never go into ACRO, not on its own, not if the user was in ACRO before, not never. This case can be easily summarized as "If WP fails for any reason, the craft will switch to the next most-autonomous mode".

This is the clearest I can argue my point, so I will stop arguing now, and will leave it up to the maintainers to do what they think is best.

wx4cb, 2018-12-01:
don't forget you need to bring some form of attention to the user in the OSD if you do this. System message maybe. not just change the "mode" text on the screen cus it might be that they don't look at that too often and by then it's too late.

shellixyz, 2018-12-02:
@skorokithakis 

> Yet you seem against a default safety hierarchy for people whose hardware don't allow them to do that?

It is a discussion about finding the best solution. Nothing have been decided yet. Personally I would be perfectly happy with your solution.

@wx4cb Sure, I brought that up earlier. Something should be displayed in the system messages OSD element and also make the aircraft beep.

skorokithakis, 2018-12-02:
Sorry, I got a bit frustrated because having the craft switch to anything manual if a mission is not loaded might very likely lead to injured people, so this is a very serious issue for me.

skorokithakis, 2019-01-31:
Please don't close this.

breadoven, 2021-11-29:
Closed by https://github.com/iNavFlight/inav/pull/5997.

closed 2021-11-29 by breadoven