iNavFlight/inav #3895: omnibus f4 pro v3    gps uart6  and  sbus uart1 conflict?

thegodfarmer, 2018-09-30:
hi i am setting up an omnibus f4 pro v3 and i cannot get my gps to work in uart6 (j10) when i plug a reciever into j7 (uart1)  but have got it (gps) to work in uart1 via j5 the 4 pin plug on the other side of the board. but this is the uart1 for the serial receiver so in ports i can only use either gps or receiver not both together. i am under the impression that there is a conflict with uart6 gps (j10) when you plug in a reciever to j7 ???
i have heard of this problem on f4 pro v2.1 but slightly different as i think uarts were assigned differently allowing use of all 3 uarts in this way?   ps the gps worked fine in j10 until a reciever is plugged into j7 the gps icon goes red and hardware health is unchecked in setup tab...... i am tearing my hair out can anybody point me in the right direction with this at all? i am quite new to inav but getting a grasp of it now. the gps is fine the reciever is fine and both i can get to work but not in the correct place on the board? ps i took a gps from a working f4 pro v2 setup and this also failed when plugged in to the pro v3
i am now sure it is the board i am having difficulty with? can the reciever be asigned to uart 3 ?? isthis an issue somebody knows how to fix?   many thanks for reading this far..

teckel12, 2018-10-01:
Why not just use the receiver on UART 1 as it's designed?

thegodfarmer, 2018-10-01:
Yeah I set it up as designed first. Sbus uart1, GPS uart6 as per the design. However when sbus is plugged in it stops GPS from working correctly?

teckel12, 2018-10-01:
@thegodfarmer I have 3 Omnibus F4 Pro FCs. All have GPS on UART 6 and  SBUS on 1. No problems at all. This is also a very popular board and configuration, so others would be having a problem.

Sounds like your FC is bad, where did you get it from?

teckel12, 2018-10-01:
@thegodfarmer My rule is I always buy FC from respectable sales outlets. Clone boards can look and smell like an original, but be wired wrong. It's usually only about a $5 savings to get it from China, and a gamble if it works at all (bad) or fails in the middle of a flight (worse).

Since we know there's something wrong with this FC, and suspect of being a clone, I'd suggest not flying with it.

Since SBUS is inverted, you can only use the inverted UART1 on this FC. Unless your Rx has pins/conversion to non-inverted. Then you could use any UART for SBUS. The other option is to run SBUS over soft serial, which can be inverted. I don't trust flying over soft serial, and the questionable connection pads for soft serial makes me less confident. But, people do use soft serial for SBUS, so it depends on your level of risk.

thegodfarmer, 2018-10-01:
Yes I think I will buy future boards from a better place like unmanned tech etc. 
However this board is kind of my first experiment with inav on a fixed wing so if I can just get it working it will be going into a small wing that holds little value to get my bearings with inav some more. 
Since I use frsky sbus and don't know anything about inverted or not then I may try soft serial.
Now let's pretend I have very little electronics knowledge but very eager to learn and I found and soldered my pad and all is wired correctly would I need to manipulate any settings in cli as I have not ventured there yet and it looks quite complicated for a beginner to work out. Or are the ports already configured just to switch on in ports tab? 
Or if that's kind of a long winded answer do you know where is best to look for the answers I will need... I thank you for taking your time to answer me as it can be frustrating. I would love to learn a bit more.

teckel12, 2018-10-01:
@thegodfarmer Actually turning on Soft Serial is just a switch.  But, to set it as inverted, full or half duplex, etc. you'll need to do that in the CLI.  Another problem with Soft Serial is where you connect it.  It's on two VERY small pads (not thru-hole connections, and in more of the center of the board).

Doing the physical connection and CLI commands are not that difficult, but if you're new to these kinds of things it may take a bit of frustration to get through.  The following documents may help:

https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md
https://github.com/iNavFlight/inav/blob/master/docs/Rx.md#sbus
https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md#smartport-sport-telemetry

thegodfarmer, 2018-10-01:
Thank you very much I will read through these tonight. Soldering is no problem for me and cli will hopefully be simple enough with correct information. I appreciate your time and effort to help me out. I will post back here if the problem is solved after I try your suggestion.

thegodfarmer, 2018-10-02:
All is now working! I decided not to try soft serial for sbus as it was going a bit off course and complicating things a bit so I sat and stared at my board for 3 hours.
Then I decided to change Rx to ppm output and changed bridge on board to ppm input for uart1. I put GPS back to uart6 and it just worked straight away??? I'm guessing the lower data of ppm is something to do with it but hey it just works. 
I checked and I got the board from a UK eBay seller and it was advertised as a genuine board but who knows EY?
I have just ordered a pro v3 from rchobby.co.uk just to see if it is different as I want to use frsky lr9 sbus eventually on my planes and I am curious aswell.
Thanks for your time and patience and I will be sure to buy genuine boards from reputable sellers from now on!

teckel12, 2018-10-03:
@thegodfarmer Sounds like the inversion circuitry on the FC isn't working. So PPM will work but SBUS won't. It could be a genuine Omnibus F4, just defective. My guess is that your replacement FC will work just fine, confirming it's a faulty FC.

thegodfarmer, 2018-12-02:
Update. A new board worked perfectly. It was a board issue. Thanks.

closed 2018-12-02 by thegodfarmer