iNavFlight/inav #3896: Cannot see motor 5 and 6 in resource list

access3424, 2018-10-01:
Im running iNav 2.0 on a kalute f4v2 and want to configure the board for a hexacopter but can only see motor 1-4 in resource. Please help

shellixyz, 2018-10-01:
AFAIK the KakuteF4v2 only has 4 outputs

access3424, 2018-10-02:
@shellixyz well thats a bummer. Is there no workaround for such a problem?

access3424, 2018-10-02:
@shellixyz you mean modify the hex file?

teckel12, 2018-10-02:
@access3424 Technically, modify the source code and compile to a custom HEX file.

access3424, 2018-10-02:
@teckel12 whats the best way to do that? I hear there are alot of options

datamann, 2018-10-02:
I just setup my dev environment using docker. It's fairly easy.

teckel12, 2018-10-02:
The other instructions are wrong in at least one place (I should probably fix that). So maybe Docker is the "easy" way, although I don't find Docker that easy.

access3424, 2018-10-02:
@teckel12 which method would you sudjest?

teckel12, 2018-10-02:
Probably Docker because other people say it's easier.

access3424, 2018-10-03:
@datamann thank you for your help! I managed to set up docker but there is little to no support as to how to modify the targets. All i want is to add motors 5 and 6. How do i do that?

teckel12, 2018-10-03:
@access3424 It would greatly help if you had a basic understanding of microcontroller resources and a bit of programming experience. It's not just a resource remapping like in Betaflight.  It may require a lot of trial and error or maybe finding someone with the same FC that could help.

access3424, 2018-10-03:
@teckel12 i do have programming experience but i cant find any support for this. do you have any material? some links?

datamann, 2018-10-03:
I have had programming as an Hobby since 1995 but I don't call my self a programmer or a developer. I do hover study this field at the moment and I have  bought some courses from Udemy (Not expensive at all). I will post the links soon... My goal is not to become as good as developers here working with INAV because they are very good. I do wish that they could create a course posted on Udemy about the INAV code structure,  I would buy it that's for sure 😊😊

datamann, 2018-10-03:
Here is the Udemy courses I have bought  and will work with. It's probably overkill but it's ok for me if I learn something. 

https://www.udemy.com/share/1004UGA0IddVZUQXQ=/

https://www.udemy.com/share/100xyeA0IddVZUQXQ=/

https://www.udemy.com/share/100jjCA0IddVZUQXQ=/

https://www.udemy.com/share/1008vcA0IddVZUQXQ=/

https://www.udemy.com/share/10052cA0IddVZUQXQ=/

https://www.udemy.com/share/1006wiA0IddVZUQXQ=/

teckel12, 2018-10-03:
@access3424 If you're looking for target-specific documentation on setting up the resources, I don't believe any such information exists.  I'd look at the target files in INAV and Betaflight and that will be a good start on what to change.  Having a little insight into microcontroller resources would also be helpful.  Look for documentation on the STM32F405

datamann, 2018-10-04:
@teckel12 It would be nice if you could take look at these Udemy courses and tell me if they will bring me in the correct direction...

closed 2018-10-05 by access3424