iNavFlight/inav #3897: INavMotors tab: Motors not showing

itswally1, 2018-10-01:
When I use  iNav to set up my NAZE board the motor tab doesn't show any of my motors. The only thing I see is a master slider. I used to see all 4 motors now only one slider. The same is true for my other Quads the were previously flashed with 1.72 ( The reason for using 1.72 is because the beeper did not work in 1.73.
Also I downloaded Inav 2.0 and have it running.

I saw a previous note on this but it was for another controller.

What do I need to do to remedy this problem?

Regards,
Don
================================= 
version 1.72 or 1.73

# dump
# INAV/NAZE 1.7.3 Sep  9 2017 / 09:59:03 (974440df)
mixer QUADX

mmix reset

servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 
smix reset

feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature VBAT
feature PWM_OUTPUT_ENABLE
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
beeper ACTION_FAIL
[... 275 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

Ralf-W, 2018-10-02:
For iNAV version up to 1.9 you have to use iNAV Configurator with version number below 2.0.
You may use Configurator 1.9.3 and 2.0 at the same time on your computer: Just keep them in different folders.

itswally1, 2018-10-03:
Thanks for the info.  I tried it and it worked. But is it a little crazy this only works on an older version of the product?
Hope you all are having fun flying.

Regards...

artistru, 2018-10-08:
just reset mixer. It works for me

closed 2018-12-21 by stale[bot]