iNavFlight/inav #3899: Support on making custom firmware

access3424, 2018-10-03:
I have a kakute f4v2 running inav. I want to convert to a hexacopter with the same board. I was told  I can make custom firmware to include motor's 5 and 6 in the resource list as they are currently not shown. The problem is that there is little support on how to edit the source files and how to compile them. I have managed to install docker but I am unable to compile as  I get the message:
 "make: *** No targets specified and no makefile found.  Stop."

Also I dont know where to begin on modifying the target files to add motor's 5 and 6.
Thanks.

datamann, 2018-10-03:
I cant help you with coding but with docker and compiling take a look at this thread. I have written in detail how to setup docker and how to compile.
https://github.com/iNavFlight/inav/issues/3896

closed 2018-10-05 by access3424

nzs001, 2018-10-20:
Hi access3424,
I got the same error message from Docker at first, although the code build should have been run alright.
It seems that the error message is not specifying all the root causes properly.
Please make sure that the Shared Drives in Settings menu of Docker is set up and checkboxed correctly.
If not, Docker drops the same error message and the user start to find the problem around the makefile but that is OK.

If you want to create custom firmware, please check this issue:
https://github.com/iNavFlight/inav/issues/3901