iNavFlight/inav #3901: How to use use pwm7 and/or pwm8 outputs for servo on Omnibus F3 for a hexacopter with Inav ?

nzs001, 2018-10-04:
Hi Everyone,

This problem I bumped into recently might be an easy one, but I could not find any solution after spending on that  about 24+ hours with different trials.

I am configuring a "OMNIBUS Betaflight F3 AIO V1.1 Flight Controller with Integrated OSD
Barometer Support SD Card" purchased from Banggood to be a hexacopter with an additional servo for a one-axis camera gimbal.
Rx is frsky on sbus (UART1), and equiped with a GPS on UART2.

Unfortunately any kind of setup with inav was unsuccessful, because it seems that output is limited to 6, although the board has 8 pwm outputs.
PWM 5&6 are shared with UART3, but I could make them free, as the sbus jumper desoldered and UART3 is not in use. Thanks to other posted issues in the community.

**PWM 7&8 are shared with i2c. I can't make the required additional 1 servo to wake up over the i2c function.**

With Betaflight I could make it with resource remapping, I have moved that one required servo to the led strip pin, but for a long time the GPS support is removed at this board in Betaflight firmwares, so this could not be a solution for me without GPS. 

Is there any simple trick like at PWM 5&6 ?

Any kind of help is welcome.
Thanks in advance,

nzs001, 2018-10-04:
Sorry, just realised that "Omnibus F3 (clones, 5 different boards) Fixed Wing NO Servo signal beyond pin 4 #3891" discusses the same issue.
Can anyone create a firmware with a disabled i2c for this board ?  I did also think about this solution but I did not even know which developing environment required for that.

datamann, 2018-10-04:
If you look at this thread https://github.com/iNavFlight/inav/issues/3896 I have listed what you can do to setups dev environment so you can try to make your own firmware.

nzs001, 2018-10-20:
Thanks for your help and guidance.

I have successfully modified the inav_2.0.0_OMNIBUS firmware to my needs to be able to use my board for a hexacopter.

If anyone else requires creating custom firmware, besides datamann's guide I can recommend the following videos:
https://youtu.be/nr5SqZvpMRI
https://youtu.be/Q8viVXarwjk
https://youtu.be/v4R-pnO4srU
These guys shares very helpful knowledge as well. 

I have created 2 versions of custom firmwares

V1
- MAX_PWM_OUTPUT_PORTS increased from 6 to 8
- no i2c port: all I2C related code sections (magnetometers, etc.) are disabled (to avoid compile errors)
- at multicopter is allows 6 motors and 2 servos and
- at flying wing it allows 3 motors and 5 servos on the PWM pins (if no UART3 is needed)
https://simaboru.hu/shop_ordered/4904/pic/others/inav_2.0.0_OMNIBUS-nzs_mod_noi2c_pwm1-8.zip

V2
- MAX_PWM_OUTPUT_PORTS increased from 6 to 8
- PPM pin is turned into a servo output for either multicopter of flying wing, so
- at multicopter is allows 6 motors and 1 servo and
- at flying wing it allows 1 motor and 6 servos on the PPM and PWM pins (if no UART3 is needed, but you are willing to use I2C on the board)
https://simaboru.hu/shop_ordered/4904/pic/others/inav_2.0.0_OMNIBUS-nzs_mod_PPM_PWMout.zip

Use them at your own risk!
This custom fw modifications were successfully tested with an "OMNIBUS Betaflight F3 AIO V1.1" board. 
Please note that I have removed the SBUS SMD jumper on my board as explained in this thread:
https://github.com/iNavFlight/inav/issues/1639
and used UART1 RX for my SBUS receiver.

These modifications migh help to others in the following issues threads:
https://github.com/iNavFlight/inav/issues/3023
https://github.com/iNavFlight/inav/issues/3891
https://github.com/iNavFlight/inav/issues/3280
https://github.com/iNavFlight/inav/issues/3896

closed 2018-10-20 by nzs001

datamann, 2018-10-21:
@nzs001 Really  Nice 😁