iNavFlight/inav #3902: Auto landing with low battery

avgorbi, 2018-10-04:
## Current Behavior
It`s look like no action on battery discharge.

## Desired Behavior
Add autolanding when battery low 

## Suggested Solution
When battery capacity bellow "Critical Capacity (%)", activate autolanding procedure.

teckel12, 2018-10-04:
In my opinion, I'm not so sure this would be helpful. First, you'd really need a current sensor due to voltage sag. Secondly, RTH is way slower than I can fly home. It also couldn't be tied to only critical capacity as distance from home would also make a huge difference.

On my Mavic the auto RTH mostly works well for two reasons.

1. It's all know fixed hardware with known characteristics
2. The battery capacity is far greater than the typical INAV quad

What I've found that works very well with INAV is just knowing what the battery capacity is while flying. This can easily be accomplished from the transmitter via telemetry. You can then use a Lua script or functions to give voice announcements of your battery level and warnings.

Basically, even if this feature existed, I don't think is turn it on as it wouldn't really be as useful as it seems. And cause more problems (like flying till the battery is at 10% critical level from 1km away, causing a power failure half way back).

avgorbi, 2018-10-25:
Of course, any function will work well only if it is properly configured.

But it will be better for mine if the copter will land softly, even though far from home, than it will fall like a stone when cut off at the speed regulator.

Hennyr, 2019-05-13:
I think this feature helps to prevent accidents

teckel12, 2019-05-13:
@Hennyr If it was realistically possible.

shellixyz, 2019-05-15:
I guess it can be done with the sag compensated battery voltage as source. I probably won't use this feature but if people want to...

teckel12, 2019-05-16:
@shellixyz There's a lot of variables, calculations, and total "fudge factor" to consider. Different kinds of RTH, different RTH speeds (not just flying, but also up and down), distance to home, wind speed, etc.  I just don't see it as being even remotely accurate.

I feel the onky way it can work is to trigger RTH at an unrealistic voltage, like 4.0 volts, just to be sure. And in that case, no one will use it.

If we had known hardware like DJI it would be possible, but I think it would need to overestimate so much it would be useless. But I'd love to be proved wrong.

shellixyz, 2019-05-16:
@teckel12 Why are you talking about RTH ? The OP only asked for landing in case the battery is almost empty to at least try to get it back in one piece. But I'm not convinced this is very useful either.

teckel12, 2019-05-16:
@shellixyz Sorry, serves me right for browsing quickly and assuming.  There were previous requests for RTH for low battery, and I just incorrectly assumed.  "auto land" is never an option for me, as my primary flying locations are near/over water, so "land" guarantees a lost/destroyed model for me.

SeanTheITGuy, 2019-09-13:
This is stale, but to add my 2c.  Either having the option to trigger failsafe or even just scaling the throttle harshly when the critical voltage level is hit would be a huge benefit.  I can't find reference to any specific feature or code in Betaflight that actions this, but I have the feeling that when I get to near the end of a pack in Betaflight, it seems to scale the throttle to where even at full, the quad is still descending.  (Though this could just be the result of voltage drop of my smaller mah packs) This would be far preferable to a sudden out of control tumble when you over discharge a pack accidentally.

michalelektryk, 2020-04-05:
I'm using this option but implemented as hack / workaround in taranis, where one channel is sending back battery voltage and FC is using it to trigger landing. Not really well done but it's the only way

DouglasKooChenSoon, 2023-09-24:
I would like to have the feature of assigning one of our transmitter channel ( any unused auxillary channel)  for activating  " Force Emergency Landing " ( i wish this to override the Failsave (which i set to RTH) ) 

The reason behind is that when i am flying long range Quad,  there is a slight chance that my Quad Battery failure ( not able to supply current ), or Motor Bearing failure ( which i couldn't check at Preflight check ),        which i know that i am in Mid air flying high altitude and not able to make it back to Home Position,      I would like to choose a "safer spot" to land using my remote video footage than crashing the drone.

i have this concern in mind, which i am afraid to fly long range ( scared of losing my pricy drone)

Please suggest me if you have any solution.  ( i wouldn't like to install parachute on my quad )

Thank you :)

MrD-RC, 2023-09-24:
This has been in INAV since 6.0

https://github.com/iNavFlight/inav/pull/8686

closed 2023-09-24 by MrD-RC