iNavFlight/inav #3903: RTH went crazy - Crash dive and crazy heights

b14ckyy, 2018-10-04:
Hi iNav Team,

I just finished my Stealth Wing Build with an F405 Wing, GPS, FPV with OSD and 3DR Telemetry. Had my first iNAV Testflight today and most things went out very well (besides some autolaunch trouble but fixed it). 

But then I made my second flight. I started Autotune to get a more stable flight and that worked out well. After that I tried manual RTH to bring it back to me and test this, before land and save the tune. 

(Issue starts at 2:26 in log) First it looked good, the plane climbed to the configured 50m height. It overshoot a bit to 54m (as per baro value in Blackbox) and then it went strange. It started to dive at a very steep angle (I'd say more than 45 degrees nose down) And I wasn't able to catch it in time by switching back to manual. So it went into the ground nose down. Fortunately the Stealth wing is very robust and just the FPV cam canopy broke off clean. I can't see any reason in the log why this happened. See this at LOG00008.TXT. After that I first forgot to disarm after impact. I put the transmitter on the ground and it lost contact because one antenna snapped off the receiver. Suddenly iNAV went into failsave and gave full throttle, not registering it is laying on the ground. I was not able to get the signal back to disarm so i ran to the plane and pulled the battery. This is also strange because failsafe status is "landed" in Blackbox.

[LOG00008.TXT](https://github.com/iNavFlight/inav/files/2447633/LOG00008.TXT)

After checking the hardware I tried again. This time i wanted to test some Waypoint Mission. The mission flying was fine again. All waypoints done. Mission was created with iNAV Configurator and I activated RTH after last WP. After the last Waypoint the plane suddenly starts to climb during flight to home point (at 2:33) and continued to climb while circling, until it stopped it by going into manual RTH (at 3:03). It started to sink while circling down to 100m and suddenly again went full throttle climbing again (3:10) and then loitered at 150m again. I accidently switched to althold/angle and back to RTH and again it started to climb (3:44). At 165m I went back to manual and brought it down manually before it goes out of sight. You can see this in LOG00012.TXT

[LOG00012.TXT](https://github.com/iNavFlight/inav/files/2447691/LOG00012.TXT)

So all modes I tested worked fine but RTH in manual and Mission flight is absolutely unpredictable. PosHold, Waypint, AltHold, Angle, Autotune, all others work well. Any Suggestions here? Find my Config Here:

[Main INAVMATEKF405SE 2.0.0 Aug 20 2018  192234 (dbdd1656a).txt](https://github.com/iNavFlight/inav/files/2447719/Main.INAVMATEKF405SE.2.0.0.Aug.20.2018.192234.dbdd1656a.txt)

stronnag, 2018-10-04:
```
$ inav_modes.rb ~/dl/LOG00012.TXT 
LOG00012.TXT: INAV 2.0.0 (dbdd1656a) MATEKF405SE
Time(s)	Elapsed(s)	State
 461.4	(   0.0)	nav_state_launch_wait (26)
 468.5	(   7.2)	nav_state_launch_in_progress (28)
 471.6	(  10.2)	nav_state_idle (1)
 487.6	(  26.2)	nav_state_poshold_3d_in_progress (7)
 520.3	(  58.9)	nav_state_cruise_2d_in_progress (30)
 521.1	(  59.7)	nav_state_cruise_2d_adjusting (31)
 522.0	(  60.6)	nav_state_cruise_2d_in_progress (30)
 522.1	(  60.7)	nav_state_cruise_2d_adjusting (31)
 522.2	(  60.8)	nav_state_cruise_2d_in_progress (30)
 522.3	(  60.9)	nav_state_cruise_2d_adjusting (31)
 530.6	(  69.2)	nav_state_cruise_2d_in_progress (30)
 534.5	(  73.1)	nav_state_waypoint_in_progress (17)
 625.6	( 164.2)	nav_state_waypoint_finished (20)
 644.5	( 183.2)	nav_state_cruise_2d_initialize (29)
 683.8	( 222.4)	nav_state_althold_in_progress (3)
 685.7	( 224.4)	nav_state_rth_head_home (10)
 690.6	( 229.3)	nav_state_althold_in_progress (3)
 690.7	( 229.3)	nav_state_idle (1)
 730.5	( 269.1)	nav_state_althold_in_progress (3)
 749.6	( 288.2)	nav_state_idle (1)
 750.4	( 289.0)	nav_state_poshold_3d_in_progress (7)
 788.4	( 327.0)	nav_state_idle (1)
 792.6	( 331.2)	nav_state_poshold_3d_in_progress (7)
 802.9	( 341.6)	nav_state_idle (1)
Disarmed by: SWITCH
```

b14ckyy, 2018-10-04:
2000us is default for fixed wing so I thought I should leave it as is. On 2 other planes (not fixed wing) and iNAV 1.9.1 i also kept it at 2000 and that worked well.

giacomo892, 2018-10-04:
[image]

No it's 1000uS default in INAV 2.0.0

Those traces doesn't look legit at all.

b14ckyy, 2018-10-04:
I used the S800 Preset because they are very similar. So yes he set it to 2000 per default ;) But you're right I'll change it to 1000 and activate filters. More than enough CPU power available. But that shouldn't have caused the issues. Because it is only RTH mode that goes crazy. If it is any interference issue, also Loiter, Waypoint flight or Cruise should have these issues. 

@stronnag is there something you want to show me with these log snippets, I don't understand or are these just notes for other devs? :D

stronnag, 2018-10-04:
Just notes ...

b14ckyy, 2018-10-04:
I think I'll fly again tomorrow. This Time I will put an SD into my googles and record the OSD to see what the plane actually does in each mode. Hope this helps. Atm I have only GoPro footage without timestamp.

giacomo892, 2018-10-05:
Try to clean up the accelerometer before. Set looptime 1K and gyro filter to 42Hz

b14ckyy, 2018-10-05:
Already done yesterday evening. Max Motor noise was at 115 Hz and I set the filters and cut-off to appropriate values. Will check the log after the flight again to see if there are any noise issues. Also I'll test RTH again if these interferences caused the issue but doubt that.

b14ckyy, 2018-10-05:
OMFG I'm such a moron… 

set nav_rth_altitude = 50000   ….   *facepalm*

This happens if you can't even count zeros.

That does not explain the sudden nose dive in the first log with the hard crash.
Also not what @giacomo892 has seen in the log: "In the first log for a reason unknown to be navTargetPos[2] is 49994 (499m) and the next step is 5145 (51m).
That's odd. navTargetPos[2] should have been 5000 (50m)"

 But it explains all the behavior after that... Also today. 

Unfortunately I had issues with my SD card to record the FPV OSD but all was fine except the permanent climb…. now I can see why…. It was in all 3 versions of my saved config 👎

b14ckyy, 2018-10-20:
I think I'll close this thread for now. No sudden dives anymore and all works as expected during my tests today.

closed 2018-10-20 by b14ckyy