iNavFlight/inav #3908: Beitian BN-180 baud speed issue

kardon18, 2018-10-07:
I have these items in use together. With iNav 2.0.0.

_Matek F411-wing FC_  (FC has no data logger on board)
_Beitian BN-180 GPS_

Settings 
- gps_auto_baud = ON
- gps_auto_config = ON
- gps_dyn_model = AIR_1G
- gps_sbas_mode = NONE
- gps_provider = UBLOX7
- inav_use_gps_velned = ON

The GPS will get a lock in under a minute (cold) every time. And most of the time it has 17 - 23 satellites, with a HDOP of less than 1.0.
My problem is that the data received from the GPS has a delay. And it gets worse the higher I set the baud rate.  
This is noticeable when viewing the Home position, Home arrow and Heading, on the OSD.
Its only partially reliable at 9600, and still has a delay of around 2 seconds.
If i raise the baud to 38400, the delay is at least 12 seconds. And 15 seconds with 57600.
But with these signal delays or data losses, it is totally unworkable for auto nav use. Even with 9600, the home loiter radius is huge. 

The GPS information being received looks fine. 
[image]

The image is with UBLOX (coordinates removed).  The delay is less with UBLOX7, but its still not a great difference.

I assume that iNav should run a GPS auto configuration. And it does seem to. Because if I enter 115200. iNav displays GPS failure on the OSD. Which is due to the GPS unit not being able to operate at that baud rate.

I have also changed to - gps_dyn_model = AIR_4G and turned the gps_velned = OFF as well.  With no difference either. 
If it had trouble getting a lock. I would assume electrical noise. But it locks on faster than most of my other GPS units.
Any idea's?

As a side issue. This also seems to be causing drifting on the Roll and Yaw axis's when in Angle modes.
The plane will drift left or right and even fly banked over. It requires full stick for a few seconds to level it up again. 
I thought this Axis drift issue was fixed. So is this a product of the GPS delay, not being able to compensate for the balance in the acceleration vector equaling Gravity in a turn?

kardon18, 2018-10-08:
Does anyone have any idea's about this?

I have spent a stack of time with it in flight testing, trying to figure these problems out. With no success.
 
I have a Matek SAM-M8Q GPS, coming in the mail. 
I do also have a second F411-wing. But I don't want to go to the all the trouble of installing it. Just on the off chance that it may be the FC. Without having some hard evidence that its at fault. 

I have also tried to set the GPS on UBX exclusively, with a higher default baud, using U-center. Just in case I might get lucky, and it would take the higher baud configuration setting. But it wasn't to be.

Is the only reason that the iNav _PORTS page_ has a UART baud selection rate, for the GPS.  Because this limits GPS_AUTO_BAUD  maximum data rate that will be used?

HFMan, 2018-10-28:
I had a maiden with a new setup, F405-Wing, Beitian BN-180 GPS.  Same issue- it's comical watching any of the GPS data on the OSD as I fly, it lags WELL behind the aircraft.  In the iNav configuration GUI, it seems to keep up with the data rate just fine from the GPS- no dropped frames, 12.5Hz most of the time.   But once I go out and fly, the data is total crap due to being way behind the airplane.  Completely useless and UNSAFE for any kind of attempt at RTH.

digitalentity, 2018-10-28:
12.5Hz is wrong. It should be 5Hz or 10Hz. Recently I see a lot of cloned Ublox modules that are experiencing this issue, but so far I don't have any on my bench which makes it impossible to debug.

stronnag, 2018-10-28:
Due to timing in the configurator, my 100% reliable BN-880 will oscillate between 8 and 12.5, averaging 10 as configured.

digitalentity, 2018-10-28:
Yes, averaging is probably causing issue on FC-side calculation of GPS update rate. Still, if it's consistently is >10Hz - this is wrong and is caused by incorrect configuration of GPS module (possibly for some reason Ublox module fails to accept all configuration stanzas INAV is sending when configuring the module).

stronnag, 2018-10-28:
Perhaps one of users experiencing this problem could help to debug using a logging tool (u-center or mwp's ublox-cli) to confirm that unwanted stanzas are being suppressed (or not). 

* From the CLI, `gpspassthrough`
* Start a GPS logger (u-center, ublox-cli). For u-center, select 'Packet Console'

Examples:
**u-center** (via Wine ...., "PC Clock" enabled in output)
```
09:19:20  [09:19:20.532]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:20.683]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:20.734]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:20.835]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:20.933]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:21.36]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:21.139]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:21.239]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:21.343]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:21.445]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:21  [09:19:21.546]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
09:19:22  [09:19:21.647]  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
```

**$ ublox-cli -p -n -d /dev/ttyACM0** (a bit verbose, sorry)
```
GPS: 2018-10-28 09:24:29.901 - Host: 2018-10-28 09:24:30.007
Data size 92b (1 7)
PVT: lat: 50.915341 lon: -1.530278 elev: 25.66 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.001 - Host: 2018-10-28 09:24:30.103
Data size 92b (1 7)
PVT: lat: 50.915340 lon: -1.530278 elev: 25.64 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.101 - Host: 2018-10-28 09:24:30.215
Data size 92b (1 7)
PVT: lat: 50.915339 lon: -1.530277 elev: 25.55 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.201 - Host: 2018-10-28 09:24:30.311
Data size 92b (1 7)
PVT: lat: 50.915340 lon: -1.530277 elev: 25.65 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.301 - Host: 2018-10-28 09:24:30.407
Data size 92b (1 7)
PVT: lat: 50.915340 lon: -1.530277 elev: 25.72 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.401 - Host: 2018-10-28 09:24:30.503
Data size 92b (1 7)
PVT: lat: 50.915341 lon: -1.530277 elev: 25.87 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.501 - Host: 2018-10-28 09:24:30.615
Data size 92b (1 7)
PVT: lat: 50.915341 lon: -1.530277 elev: 25.91 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.601 - Host: 2018-10-28 09:24:30.711
Data size 92b (1 7)
PVT: lat: 50.915340 lon: -1.530275 elev: 25.82 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.701 - Host: 2018-10-28 09:24:30.807
Data size 92b (1 7)
PVT: lat: 50.915340 lon: -1.530275 elev: 25.84 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.801 - Host: 2018-10-28 09:24:30.903
Data size 92b (1 7)
PVT: lat: 50.915341 lon: -1.530275 elev: 25.93 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:30.901 - Host: 2018-10-28 09:24:30.999
Data size 92b (1 7)
PVT: lat: 50.915341 lon: -1.530274 elev: 26.01 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:31.001 - Host: 2018-10-28 09:24:31.111
Data size 92b (1 7)
PVT: lat: 50.915340 lon: -1.530273 elev: 25.88 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:31.101 - Host: 2018-10-28 09:24:31.207
Data size 92b (1 7)
PVT: lat: 50.915338 lon: -1.530272 elev: 25.79 acc(h/v): 6.6/11.8
sats: 4, fix 3
GPS: 2018-10-28 09:24:31.201 - Host: 2018-10-28 09:24:31.303
```
Just PVT (UBLOX-7 setting) at 10Hz. ublox-cli in particular shows the timing (and the PC appears 0.1s fast of the GPS time ....).

HFMan, 2018-10-28:
What Baud rate is iNav setting the GPS to?  I did the gpspassthrough stuff
last night and although I did not see any extra messages, I saw the
timestamp of the PVT packets slowly but surely lag behind PC time.  I also
noticed that if I forced the baud rate to 115200 via a U-Blox CFG command
(and also turned off auto-baud on iNav and set gpdbaud to 115200), then
this packet timestamp kept up with PC time.

digitalentity, 2018-10-28:
Switch from `UBLOX7` to `UBLOX` protocol and check if problem persists. If it is resolved - we have found another kind of problematic GPS modules

stronnag, 2018-10-28:
In my case, the GPS is at 38400 baud (from the iNav port setting); `gpspassthrough` does not change that. On the BN-880 I hadn't noticed any elapsed time related change in delta between PC and GPS time (not that I was looking for it). I'll leave it to run for a while and see.

HFMan, 2018-10-29:
Here is what I have found out.  The only way to get a consistent GPS connection at startup is at 38400 with autobaud turned off.  I have to make sure I go into UCenter and send a cfg-cfg message to save the baud rate at 38400 and save the current config to BB Ram and Flash.

If I have autobaud **enabled** in iNav (with the serial port set to 38400), it never can get synced up with the gps on a cold start of the FC.  After this first cold start failure, iNav flashes the GPS icon Red after several seconds of communication, but finally iNav does get data flowing and lights up the GPS icon (several seconds later)  After this rough start, if I enable gpspassthrough and check in U-Center almost always finds the GPS running at 9600 baud with NMEA messaging.  I found in further testing that after this cold start, if I simply reboot the FC (going to CLI, then exiting), sometimes iNav can get things going better using UBX with PVT only messages, but often the baud rate ends up at 9600 baud.  After another reboot of the FC, it may get closer- generally it takes 3-4 reboots of the FC (but not the GPS, it stays powered up the entire time)  to get things running properly at 38400 with UBX PVT messages only.

This is always the case with autobaud turned on, even if the serial baud rate matches what is currently stored in the M8N.  If, however, I **turn off autobaud** and the M8N is configured for 38400 as is iNav, then every single boot, whether a cold start of the FC or a simple reboot- it immediately hooks up to the GPS and starts flowing data (yes, I have autoconfig enabled for the GPS for all these tests).  Obviously if successfully running at 38400 there is no lag in the GPS data received.

So it seems to be that with autobaud enabled, iNav is somehow trying to communicate with the GPS and is getting messed up.  I think all the messages that the GPS is already spewing out may be messing up iNav on a cold start of the FC with autobaud enabled.  It's a struggle- I have tried literally dozens of permutations of this and going into UCenter to verify what I see.

digitalentity, 2018-10-29:
Nice find. Looks like baud rate is messing up some of GPS modules, I'll look into it.

HFMan, 2018-10-29:
I just tried the patch (F405-SE).  It still gets messed up if we are in a cold boot and the GPS has lost its configuration data (I think the BN-880 has a teeny cap that seems to die in just a hour or so).  Cold starts when the BN-880 still has power in the super cap is working fine with autobaud enabled.

digitalentity, 2018-10-29:
@HFMan can you try this hex? 

I've done complete overhaul of UBLOX driver (only that driver so far, NMEA and the others are disabled completely). GPS may take a bit more time to initialize here, but if it works in general we can tweak it.

[inav_2.1.0_MATEKF405SE.hex.zip](https://github.com/iNavFlight/inav/files/2526966/inav_2.1.0_MATEKF405SE.hex.zip)

HFMan, 2018-10-29:
@digitalentity - So far so good.  I can seem to do a cold start of the FC and it does take a bit to get sorted.  After a bit the GPS icon turns red, then finally turns blue.  10hz messages, and enabling gpspassthrough validates the communication is set at the desired 38400baud.  I'll test more, and we'll see how it does after sitting powered off all night.  Thanks for the effort!

digitalentity, 2018-10-30:
Great news. I'll wait for your next test and look into making GPS bringup faster.

HFMan, 2018-10-30:
@digitalentity - Still looks good.  Cold start this morning and it got configured properly.

digitalentity, 2018-10-30:
Cool! Next step would be to clean up and fix the rest of the GPS drivers.

jessejgm, 2021-06-13:
What's this program?

PR #3974 (2018-10-31): GPS code overhaul (backport)
Backport of #3962 to 2.0.0-stable to create a 2.0.1-maintenance release.

Fixes #3908, #3556